innoveit/react-native-ble-manager #761: How to decode BleManager.read response ?

mvmmarcus, 2021-07-27:
I am trying to do this:

[image]

but the response persists encoded...

**Revelant Infos:**
serviceUUID: 1808 => Glucose Service
characteristUUID: 2A08 => Date Time Characteristic

**PS**
Im getting the same encoded response to anyone characterist that i tried

SaltedBlowfish, 2021-07-29:
The data is a byte array. You must know what each byte represents, then you can "decode" however you want.

That's a fundamental thing you have to determine for yourself and your own data types. There is not one magical "decoding" process because in computers, variables are all stored in binary and it's up to you to know how to "decode" them. Same with BLE since the data is transmitted in bytes.

Quick Example:

You might have 2 bytes that are part of the same 16 bit unsigned integer value you need from your BLE device. Let's say `packet[0]` and `packet[1]` refer to their bytes, where `packet[0]` refers to bits `0 - 7`, `packet[1]` refers to bits `8 - 15`. You can:

```
// the value we are "transmitting" over BLE is 14537, represented in 2x8 bit values as `[56, 201]`
const incomingPacket = [56, 201]

// we are shifting the first byte left by 8 bits (size of one full byte) and adding to the unshifted remaining byte
const incomingPacketDecoded = incomingPacket[0] << 8 + incomingPacket[1]

console.log(incomingPacketDecoded) // 14537
```

If you have ASCII char values in the byte array, you can do this:

```
const incomingPacket = [65, 66, 67] // A, B, C respectively

// send each byte-represented character through JavaScript's internal method for decoding from char codes, then join to form a full string
const incomingPacketDecoded = incomingPacket.map(i => String.fromCharCode(i)).join()

console.log(incomingPacketDecoded) // "ABC"
```

Hope this helps. It's outside the purpose of this library to decode the data because there are an uncountable number of ways to structure the data from a BLE device. You could even mix chars, signed ints, unsigned ints, floats, etc. all in one packet)

mvmmarcus, 2021-07-29:
> Os dados são uma matriz de bytes. Você deve saber o que cada byte representa, então você pode "decodificar" como quiser.
> 
> Isso é uma coisa fundamental que você deve determinar para você e seus tipos próprios de dados. Não existe um processo mágico de "decodificação" porque nos computadores, as variáveis ​​são todas armazenadas em binário e cabe a você saber como "decodificá-las". O mesmo com BLE, pois os dados são transmitidos em bytes.
> 
> Exemplos rápidos:
> 
> Você pode ter 2 bytes que fazem parte do mesmo valor inteiro não assinado de 16 bits que você precisa do seu dispositivo BLE. Digamos `packet[0]`e `packet[1]`consulte seus bytes, onde `packet[0]`se refere a bits `0 - 7`, `packet[1]`se refere a bits `8 - 15`:
> 
> ```
> // the value we are "transmitting" over BLE is 14537, represented in 2x8 bit values as `[56, 201]`
> const incomingPacket = [56, 201]
> 
> // we are shifting the first byte left by 8 bits (size of one full byte) and adding to the unshifted remaining byte
> const incomingPacketDecoded = incomingPacket[0] << 8 + incomingPacket[1]
> 
> console.log(incomingPacketDecoded) // 14537
> ```
> 
> Se você tiver valores ASCII na matriz de bytes, você pode fazer isso:
> 
> ```
> const incomingPacket = [65, 66, 67] // A, B, C respectively
> 
> // send each byte-represented character through JavaScript's internal method for decoding from char codes, then join to form a full string
> const incomingPacketDecoded = incomingPacket.map(i => String.fromCharCode(i)).join()
> 
> console.log(incomingPacketDecoded) // "ABC"
> ```
> 
> Espero que isto ajude. Está para o propósito desta biblioteca decodificar os dados porque há um número incontável de maneiras de estruturar os dados de um dispositivo BLE. Você pode até mesmo misturar chars, ints assinados, ints não assinados, floats, etc., tudo em um pacote)

Thank you so much for the answer. Now i can understand what format the data is received.
That is my solution to anyone that has the same question:

[image]

closed 2021-07-30 by marcosinigaglia