innoveit/react-native-ble-manager #762: Scan issue

SafdarSikander, 2021-07-27:
### Version 7.6.0

Tell us which versions you are using:

- react-native-ble-manager v7.6.0
- react-native v0.62.2
- iOS/Android v.Android

### Expected behaviour

Start scan functionality a couple of times and it stops when timeout but after 2 or 3 times it stuck in scanning and not connect to Bluetooth also.

### Actual behaviour

It should stop the scan on timeout 

### Steps to reproduce

1. call scan functionality on button

Goldtek, 2021-08-06:
react-native-ble-manager v7.4.2
react-native v0.63.3
IOS

joy-mollick, 2021-09-16:
It is not working perfectly now ,so many bugs in library. Not a good choice for bluetooth connection . There are other packages you can use .

Goldtek, 2021-09-17:
which do you suggest?

dominicbar, 2023-03-15:
Hey did you find a solution for this?