innoveit/react-native-ble-manager #770: When I disconnect a device, I can't rescan to the device on Android 8.0

linfeizhang, 2021-08-25:
When I disconnect, I can't rescan to the device on Android8

- react-native-ble-manager v7.5.0
- react-native v0.63.4
- Android v8

Steps to reproduce:
1. BleManager.scan()
2. Discover attributes (Services, characteristics, etc) for device.
3. BleManager.connect()
At this point, if I read / write to characteristic, the library acts as expected and performs the procedure.
3. BleManager.disconnect()
4. BleManager.scan()
Unable to scan the device again,It's not actually disconnected from that device.

Please tell me how to solve this problem?

linfeizhang, 2021-08-30:
> ble code

I don't have any error messages, I'm showing a successful disconnect.But, in fact, there is no disconnect.

marcosinigaglia, 2021-08-30:
@linfeizhang probably is a device bug.

LazloN5, 2021-12-06:
Any updates on this?

Dustin-PromenadeSoftware, 2021-12-09:
Running into a similar issue. Any updates on how you resolved this?

LazloN5, 2021-12-10:
What helped for me is `removePeripheral` : [https://github.com/innoveit/react-native-ble-manager#removeperipheralperipheralid-android-only](url)

I'm not sure if this is **_the_** way to go, but it fixed my issue (for now)

Dustin-PromenadeSoftware, 2021-12-10:
@LazloN5 
Do you by chance have a snippet of what you did? I tried a few combinations but seems to not disconnect properly still.

dsnarwariya0102, 2022-12-02:
I don't have any error messages, I'm showing a successful disconnect.But, in fact, there is no disconnect.Because after again rescan then device not show in the list

benedicksteve, 2023-11-29:
I also got similare issue with Pixel 8 on Android 14, first scan attemps work fine, I'm able to connect, then if I disconnect and trigger a scan again I will not have any "BleManagerDiscoverPeripheral" signal for this peripheral. 

I tried calling removePeripheral on disconnection but didn't change anything. Look like peripheral is cached somewhere but didn't find out how to clean this cache to be able to discover it again.

I tried tu run a ble scan via nrfConnect in parallele and my peripheral was in reach. Also find out that triggering a scan with nrf in parallele allowed me to find my device in my app again.

As work around I loop on discovered peripherals before my new scan (getDiscoveredPeripherals) to get my device information as my peripheral is still present via this function, but would be better to get it via scan again to make sure he's still at reach.

anwarhamr, 2024-07-11:
@marcosinigaglia can you point me to the ticket/change that fixed this?  We are on an old version and cannot upgrade.  I'm wondering if I can add this back into our fork.