innoveit/react-native-ble-manager #773: BleManager.connect - most likely the callback was already invoked. Args: '[]'

Heltonphg, 2021-09-15:
- react-native-ble-manager: ^7.4.1
- react-native v: 0.63.4
- Android v: 11

### Expected behaviour

Expected there is a connection

### Actual behaviour
Out of 10 connection attempts, 8 happen perfectly, but in 2 or 3 this exception occurs:
```
Invariant Violation: No callback found with cbID 3135 and callID 1567 for  BleManager.connect - most likely the callback was already invoked. Args: '[]'

```
### function connect

```
 let uuid = priority_list[0].uuid;
    try {
      dispatch(BluetoothActions.set_start_process(true));
      if (Platform.OS == 'ios') {
        BleManager.connect(uuid).then(() => {});
      } else {
        await BleManager.connect(uuid);
      }
      await BleManager.retrieveServices(uuid, ['ffe0']);
      await BleManager.startNotification(uuid, 'ffe0', 'ffe1');
    } catch (e) {
    ......
    }
```

Shirsendu-Sarkar, 2021-10-05:
`Using following versions
"react-native-ble-manager": "^7.6.3",
"react-native": "0.64.2",

  iam also getting same error

  var peripheralId = "B0:B1:13:85:DD:88"
  var retries=10
  var timeout = 500
  const connectTimeout = React.useRef(null);

  const attemptConnection3 = async (id, i, timeout) => {
    console.log(`connection attempt on ${id} is ${i}...`);
    try {
      await new Promise(async (resolve, reject) => {
        connectTimeout.current = setTimeout(
          () => reject('connection timeout'),
          timeout,
        );

        await BleManager.connect(id);

        if (connectTimeout.current) {
          clearTimeout(connectTimeout.current);
          connectTimeout.current = null;
          resolve();
        } else {
          console.log('else attempt');
          reject();
        }
      });
    } catch (err) {
      clearTimeout(connectTimeout.current);
      connectTimeout.current = null;
      console.log(`Could not connect to device ${id} due to err: ${err}`);
    }
  };

  const bleConnect = async (peripheralId, retries, timeout) => {
    var connected = false;
    for (let i = 1; i <= retries; i++) {
      await attemptConnection3(peripheralId, i, timeout);
      connected = await BleManager.isPeripheralConnected(peripheralId);
      if (connected) {
        break;
      }
    }
    return connected;
  };

  bleConnect(peripheralId, retries, timeout); //calling here
  

  `

soltesza, 2021-10-13:
I'm encountering this too when trying to reestablish a connection to a peripheral. The same code works fine on iOS. I'm using Android v7.1.1

closed 2022-02-02 by marcosinigaglia

eastpiger, 2022-03-04:
Same issue here. Not shown on my android 12 device about almost every time on android 10 device.

ManigandanRaamanathan, 2022-09-06:
I'm facing same issue on android 12. Even try catch is not helping

soumitaTechnoexponent, 2022-12-28:
I'm facing same issue on android 12. Not getting any solution for it.

Tenfleques, 2023-03-07:
Resolved this by storing the promise in a global array, pop it out in  then or catch or always.

Before calling ble.connect, I check if this array is 0 length.

If 2 or more ble.connect promises run, they somehow clash and raise this exception