innoveit/react-native-ble-manager #775: Not working package , not compatible with new react-native .

joy-mollick, 2021-09-16:
It is unnecessary package, it's not working !!! How do people use this,I want to know.Only scanning no result , trying with several model phones android 9,10,11 restart my phone several times .On and of my Bluetooth but there is no devices found !! This package is not compatible with new rect-native ,It is better to close this package.  Otherwise people will struggle with this ..

Poojamok, 2021-09-20:
I think you did not install this library in correct way that's why you could not get anything else your code have an issue, for me  its working like charm and I am recently using this library. thank you

whatdtech, 2021-09-21:
Hello  
joy-mollick . This library is working perfectly on android 11 and RN 0.65.2. You should follow the documentation perfectly and for scanning to work location should be enabled. If you know the mac-id of the device which needs to be connected directly use BleManager.connect("mac-id").

closed 2021-09-21 by marcosinigaglia