innoveit/react-native-ble-manager #776: NativeModules.BleManager is null

Pifoupif, 2021-09-21:
### Version

- react-native-ble-manager v7.5.0
- react-native v0.65.1
- iOS/Android IOS v14.4.2 (haven't checked on Android)

### Expected behaviour

I should be able to instantiate a `new NativeEventEmitter` with the `NativeModules.BleManager` as argument

### Actual behaviour

I'm at the very beginning of implementing the lib and I get the following error:
`new NativeEventEmitter() requires a non-null argument`

My code is as follow:

```
import BleManager from 'react-native-ble-manager'
import { NativeModules, NativeEventEmitter } from 'react-native'

const BleManagerModule = NativeModules.BleManager
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule)
```

When console logging `BleManagerModule`; it's indeed `null`

### Stack trace and console log
`2021-09-21 15:10:12.916015+0200 rider[824:161950] [javascript] 'Invariant Violation: `new NativeEventEmitter()` requires a non-null argument.', '\nApp@http://10.20.1.62:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=org.reactjs.native.bloom.rider:108073:48\nRCTView\nView\nRCTView\nView\nAppContainer@http://10.20.1.62:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=org.reactjs.native.bloom.rider:75046:36\nrider(RootComponent)@http://10.20.1.62:8081/index.bundle?platform=ios&dev=true&minify=false&modulesOnly=false&runModule=true&app=org.reactjs.native.bloom.rider:82263:28'`

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2021-10-01:
Hi @Pifoupif , is an installation problem.

closed 2021-10-01 by marcosinigaglia

ChenWenWu223, 2023-01-05:
This problem may be caused by ios not installing react-native-ble-manager correctly. You can try the following commands:
`cd ios && pod install`