innoveit/react-native-ble-manager #777: Notification error Set notification failed

AMINDRH91, 2021-09-22:
### Version

-     "react-native-ble-manager": "7.6.1",
-    "react-native": "0.64.0",
- Android :v 9 with android studio Android Studio Arctic Fox | 2020.3.1 Patch 2

Build #AI-203.7717.56.2031.7678000, built on August 27, 2021
Runtime version: 11.0.10+0-b96-7281165 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
macOS 11.5.2
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 4
Registry: external.system.auto.import.disabled=true

### Expected behaviour

I work for a startup
Which sends an array string to the requests sent via the device via Bluetooth
It used to work great for both Android and iOS, but now I get it when I use " BleManager.startNotification "  Error  " Notification error Set notification failed " 

I'll share my code : 

`const App = () => {
  const [isScanning, setIsScanning] = useState(false)
  const peripherals = new Map();
  const [list, setList] = useState([]);

  const startScan = () => {
    if (!isScanning) {
      BleManager.scan([], 3, true).then((results) => {
        console.log('Scanning...');
        setIsScanning(true);
      }).catch(err => {
        console.error(err);
      });
    }    
  }

  const handleStopScan = () => {
    console.log('Scan is stopped');
    setIsScanning(false);
  }

  const handleDisconnectedPeripheral = (data) => {
    let peripheral = peripherals.get(data.peripheral);
    if (peripheral) {
      peripheral.connected = false;
      peripherals.set(peripheral.id, peripheral);
      setList(Array.from(peripherals.values()));
    }
    console.log('Disconnected from ' + data.peripheral);
  }

  const handleUpdateValueForCharacteristic = (data) => {
    console.log('Received data from ' + data.peripheral + ' characteristic ' + data.characteristic, data.value);
  }

  const retrieveConnected = () => {
    BleManager.getConnectedPeripherals([]).then((results) => {
      if (results.length == 0) {
        console.log('No connected peripherals')
      }
      console.log(results);
      for (var i = 0; i < results.length; i++) {
        var peripheral = results[i];
        peripheral.connected = true;
        peripherals.set(peripheral.id, peripheral);
        setList(Array.from(peripherals.values()));
      }
[... 200 lines cut ...]
    paddingHorizontal: 24,
  },
  sectionTitle: {
    fontSize: 24,
    fontWeight: '600',
    color: Colors.black,
  },
  sectionDescription: {
    marginTop: 8,
    fontSize: 18,
    fontWeight: '400',
    color: Colors.dark,
  },
  highlight: {
    fontWeight: '700',
  },
  footer: {
    color: Colors.dark,
    fontSize: 12,
    fontWeight: '600',
    padding: 4,
    paddingRight: 12,
    textAlign: 'right',
  },
});`

AMINDRH91, 2021-09-22:
@marcosinigaglia Please save my life :((((
Thank You

AMINDRH91, 2021-09-22:
[image]

marcosinigaglia, 2021-10-01:
Hi @AMINDRH91 , are you sure that the characteristic can notify?

TyshkevichSergey, 2021-10-04:
Have close issue, but in adb logs seems like all is ok, but I catch error like set notificatoin failed for <Char UUID>

AMINDRH91, 2021-10-05:
No it's Open issue i did not get answer yet @TyshkevichSergey

AMINDRH91, 2021-10-05:
Yes, It's working great on the IOS , but in the android send notification not work

mikcox, 2022-04-25:
Any progress on this?

I'm having a similar issue... It works fine in iOS, but all Android versions (we've tested 9 and 12 at this point) are giving the same issue as above.

To reiterate the problem I believe we're facing:
- We subscribe to notifications on characteristic A
- We try and read a value from characteristic B
- We successfully get the value from characteristic B
- We see an error with the notification `Failed to set client characteristic notification for UUID-for-characteristic-B`

enchorb, 2022-12-07:
Anyone figured out why this is happening? Bump @marcosinigaglia

mutablestudio, 2023-03-21:
I believe you need to add the listener after you start notifications for all characteristics:

```  
setNotifyOnUpdate = async (): Promise<boolean> => {
      await this.bleManager.startNotification(this.peripheralId, <service>, <characteristicA>)
      await this.bleManager.startNotification(this.peripheralId, <service>, <characteristicB>)
     // only do this once you've started notifications for all the characteristics you want...
      bleManagerEmitter.addListener(
        "BleManagerDidUpdateValueForCharacteristic",
        this.handleUpdateValueForCharacteristic
      )
  }
```

Your logic to handle notifications would be in `handleUpdateValueForCharacteristic()`.

closed 2023-12-05 by marcosinigaglia