innoveit/react-native-ble-manager #782: Example doesn't work

ksetrin, 2021-09-28:
Tried out example https://github.com/innoveit/react-native-ble-manager/tree/master/example
But I can't see any peripherals on the screen

[image]

But what I see in the Xcode logger

[image]

Looks like it works, but how I can handle all peripherals in the my react component?

gfmoore, 2021-09-29:
Heyup,  I've just been working with this code on android. The example sort of works for me and I've adapted it as I learn how this ble stuff works. I'm using a wearable garmin heart rate monitor at the moment for my tests.

What I do find is that scanning is a bit of a hit and miss affair. If I log what is being found when the 
```
const handleDiscoverDevice = (device) => {
    console.log(device.name)
}
```
is run (I use device rather than peripheral ;) ) is that sometimes no devices are found, sometimes a load of device.name is null is found, occasionally it finds my hr monitor and sometimes I think I even get it twice.

The phone, hr are all sitting beside me, so why it doesn't just find things each time is puzzling. 

Btw, stupid question, but are your bt devices connected via bluetooth to your phone.

Anyway, here's my cut down code if you want to give it a whirl: NB It's still just test code as I learn stuff...

```
/**
* Using Bluetooth
 */

import React, { useState, useEffect } from 'react'
import { Platform, PermissionsAndroid, LogBox, NativeModules, NativeEventEmitter, StyleSheet, View, Text, TouchableOpacity, Scrollable, FlatList } from 'react-native'

import BleManager from 'react-native-ble-manager'
const BleManagerModule = NativeModules.BleManager
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule)

//https://stackoverflow.com/questions/41146446/get-rid-of-remote-debugger-is-in-a-background-tab-warning-in-react-native/54392003#54392003
//check priority box in debugger (top left, to the right)
LogBox.ignoreLogs(['Remote debugger'])

const App = () => {

  const devicesfound = new Map()    //tried new Set()
  const [devices, setDevices] = useState([])

  //initialise ble manager
  useEffect( () => {
    //handle permissions
    if (Platform.OS === 'android' && Platform.Version >= 23) {
      PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION).then((result) => {
        if (result) {
          console.log("Permission is OK");
        } else {
          PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION).then((result) => {
            if (result) {
              console.log("User accept");
            } else {
              console.log("User refuse");
            }
          });
        }
      });
    }

    BleManager.start({ showAlert: false }).then(() => {
      console.log("GM: BLE initialised")
    })

    bleManagerEmitter.addListener('BleManagerStopScan', handleStopScan)
    bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', handleDiscoverDevice)  //calls this if device discovered during scan
  }, [])

  const startScan = async () => {
    devicesfound.clear()
    setDevices([])
    try {
      console.log('GM Scanning...')
      await BleManager.scan([], 3, false)   //these options were supposed to stop duplicates??, false, {numberOfMatches: 1, matchMode: 1, scanMode: 1, reportDelay: 0} ) )  
    }
    catch(e) {
      console.log('GM: Error scanning ' + e)
    }
  }

  const stopScan = async () => {
    try {
      await BleManager.stopScan()
      console.log('GM: Stop scan')
      setDevices(Array.from(devicesfound))
    }
    catch(e) {
      console.log('GM: Error in stopping scan ' + e)
    }
  }

  const handleStopScan = () => {
    console.log('GM: Stopped scanning')
  }

  //if device discovered during scan
  const handleDiscoverDevice = (device) => {
    console.log(device.name)
    if (device.name !== null) {
      console.log('adding')
      devicesfound.set(device.id, device);
      setDevices(Array.from(devicesfound.values()));
    }
  }

  const listDevices = () => {
    devices.forEach(d => {console.log(d)})
  }

  const selectDevice = (d) => {
    console.log('here', d)
  }

  return (
    <View>
      <Text style={ styles.text }>Bluetooth</Text>

      <View style={ styles.scanContainer }>
        <TouchableOpacity style={ styles.button } onPress={startScan}>
          <Text>Scan</Text>
        </TouchableOpacity>
        <TouchableOpacity style={ styles.button } onPress={stopScan}>
          <Text>Stop Scan</Text>
        </TouchableOpacity>
      </View>

      <TouchableOpacity style={styles.button} onPress={listDevices}>
        <Text>List devices</Text>
      </TouchableOpacity>
      
      {/* <Scrollable> */}
      <FlatList
        data={devices}
        keyExtractor={devices => devices.id}
        renderItem={({ item }) => { 
          return <TouchableOpacity onPress={selectDevice} style={'styles.devicerow'}><Text style={styles.devicerowtext}>{item.name}</Text></TouchableOpacity> 
        }}
      />
     {/* </Scrollable> */}

    </View>
  )
}

const styles = StyleSheet.create({
  scanContainer: {
    flexDirection: 'row'
  },
  button: {
    marginLeft: 20,
    marginTop: 20, 
    width: 150, 
    height: 70, 
    backgroundColor: 'lightgreen',
    alignItems: 'center',
    justifyContent: 'center',
  },
  text: {
    marginLeft: 20,
    marginTop: 20, 
    fontSize: 20,
  },
  devicerow: {
    backgroundColor: 'lemonchiffon',
    width: 200,
    height: 40,
    borderWidth: 1,
    borderColor: 'grey',  
  },
  devicerowtext: {
    marginLeft: 20,
    marginTop: 20, 
    fontSize: 20,

  },
})

export default App

```

kangevall, 2021-09-30:
Ksetrin, we are seeing similar issues with the latest releases. Do you see the same issues if you run version 7.4.2?

ksetrin, 2021-09-30:
@kangevall 
sorry, I already switched to react-native-ble-plx

ksetrin, 2021-09-30:
@gfmoore 
Thanks a lot for your example.
I'm sure it will be useful for many others

nitrique, 2021-10-01:
> @kangevall sorry, I already switched to react-native-ble-plx

This lib don't work with XCode 13 as it is not maintained anymore

marcosinigaglia, 2021-10-01:
Hi guys, if there is a problem with the latest release use the previous one, it's open source, you can check the problem and make a PR.

closed 2021-10-01 by marcosinigaglia