innoveit/react-native-ble-manager #784: BLE events are not working on iOS release build

nitrique, 2021-09-30:
Hi,

I have a mistake with release build only,

### Version

Tell us which versions you are using:

- react-native-ble-manager v7.6.2
- react-native v0.64.2
- iOS 15

### Expected behaviour

Expected to receive events

### Actual behaviour

Note that it is working on development environment, but not on release one, using scheme "release"

### Steps to reproduce

1. Go to XCode > Product > Scheme > Edit scheme > Build configuration = Release (instead of debug)
2. Build the app
3. When app is running, no event is triggered on JS side, but the log show `Discover peripheral: peripheral name`

If I disable the "if" on discover peripheral, it show 

```
[native] Sending `BleManagerDiscoverPeripheral` with no listeners registered.
```

But if I query registered listener count with `bleManagerEmitter.listenerCount('BleManagerDiscoverPeripheral')` , it show 1 listener registered.

I've tried with  the following: 

- Start before addListener
- Listener before start
- Many start, many listener

In any case, start and scan are not on the same window, start is done on app homepage, but scan is done on parameters page, so the module is warmed up a long time before scanning (note that the native module find multiple peripherals).

Home someone has the answer, I'm stuck since days.

Thanks,
Regards.

Nicolas

nicolasdevienne, 2021-10-01:
Hi, try uninstall v7.6.2 and install v7.5.0

nitrique, 2021-10-01:
> Hi, try uninstall v7.6.2 and install v7.5.0

You are right it works !

nicolasdevienne, 2021-10-01:
v7.5.0 seems to be the latest good version. I worked with this version for my previous project. I start a new project this morning with v7.6.2 but events are not working...

closed 2021-10-01 by marcosinigaglia