innoveit/react-native-ble-manager #785: startNotification not working

wstuckmeyer, 2021-10-03:
-iOS and Android
 - react-native-ble-manager version: 7.6.3
 - react-native version: 0.63.2
 
**Describe the bug**
I'm trying to retrieve an HRV reading from a device. After connecting and retrieving services I'm running `startNotification` on the Heart rate service characteristic which is notify only. My `BleManagerDidUpdateValueForCharacteristic` is never triggering because the device never starts notifying. 

However, when I connect to the device with a different app from the device manufacturer and then go back to my app `startNotification()` is working correctly and `BleManagerDidUpdateValueForCharacteristic` begins receiving data.

My guess is the the `2902` descriptor which controls if the device is notifying or not is never being changed, and I can't fine a way to write or update this.

Steps to reproduce the behavior:
1. Connect to peripheral
2. Retrieve services
3. StartNotification for characteristic
4. Set listener for changes

**Expected behavior**
App gets heart rate data from the device.

Here's the function for context. Also to reiterate this code works correctly AFTER connecting device to manufacturers app first, but not when this code is run first by itself after a device reset.

```
 BleManager.connect(peripheral_id)
           
                this.setState({
                    connected_peripheral: peripheral_id
                });

                // retrieve the services advertised by this peripher
                
                BleManager.retrieveServices(peripheral_id).then(
                    (peripheralInfo) => {

             
                        let service = "180D"
                        let characteristic = "2A37"
                       
                        
                        BleManager.startNotification(peripheral_id , service, characteristic).then((data, whatever) => {
                            // Success code
                           
                            console.log("Notification started");
                            
                            bleManagerEmitter.addListener(
                                "BleManagerDidUpdateValueForCharacteristic",
                                ({ value, peripheral, characteristic, service }) => {
                                console.log(value, peripheral, characteristic, service)
                                    // Convert bytes array to string
                                    // const data = bytesToString(value);
                                    // let allData = this.state.data
                                //    console.log(value)
                                    // let timestamp = value.slice(1, 6).reverse()
                                    // let converted = moment.unix(timestamp).format("MM DD YYYY")
                                    // let hrv = value[9]
                                    // let heartRate = value.slice(7,8).reverse() /60
                                    // let newData ={
                                    //     Time:converted,
                                    //     HRV:hrv,
                                    //     HR:heartRate
                                    // }
                                    // if(allData.length <= 10){
                                    //  allData.push(newData)   
                                    // }
    
                                    // this.setState({data:[newData]})
                                }
                            )
                        }).catch((error) => {
                            // Failure code
                            console.log(error);
                        })
                    })
                    .catch(err => console.log(err)

            })
            .catch((error) => {
                console.log(error)
               
            })
            .catch(err => console.log(err))
```

**Additional context**
Add any other context about the problem here.

gfmoore, 2021-10-04:
I had this problem when developing the last few days. Not only that but using nrf connect app or light blue app also gave inconsistent results from a garmin chest heart rate monitor. My mobile worked on the apps then after trying to develop my app didn't work. Changed battery, tried all sorts of things, nowt. Then I tried my tablet using the light blue and nrf and it worked fine. Back to my app nothing. Drove me mad. Back to light blue nothing, what the heck! Then the phone would work... 

But then my app started working. 

The only thing I can figure out is that if something has started the notifying process then it needs to be stopped before using it on another app?  I haven't tested this, but it's worth a look. Also my code, that is working (sort of) is on https://github.com/gfmoore/BLETest 

If you want to try it rather than looking at code, then let me know and I'll put some notes together on how to use it.

Hope this helps. This bluetooth stuff is a nightmare to use.

gfmoore, 2021-10-04:
I've refactored my BLETest app a little:

It stores a list of saved peripherals in async storage with a key starting with 'BLE'

You can press the bluetooth button to scan - there is some console log to show what it is showing. It is weird as sometimes it doesn't see anything, sometimes it does, sometimes its just nulls. Click the BT icon and click it again. Eventually it should display a list of found BT peripherals.
press on the peripheral to add it to the list of saved peripherals (These should be persistent). )You can click on the BT icon to remove the scan results.

Press on a peripheral in the saved peripherals list to connect - now this is hardcoded for my Garmin HRM because I don't know what to do next? However, it should start showing the heart rate.

Press on the now blue Peripheral to stop the notification. 

(This is basically just so you can follow the code - it is a work in progress, but maybe it helps - but it DOES work.)
(Perhaps I should post the code here, but it is a little long?)

TyshkevichSergey, 2021-10-04:
have same issue,  with error Set notification failed for /*char UUID*/ token writing

wstuckmeyer, 2021-10-04:
I'm closing this issue. The problem was bad documentation for the device. I needed to write to one of the characteristics first before it would begin recording data. This library works as expected!

closed 2021-10-04 by wstuckmeyer

TyshkevichSergey, 2021-10-05:
> have same issue, with error Set notification failed for /_char UUID_/ token writing

also found solution. Checked in services after retrieveServices that characteristic wasnt notifying on device side

fukemy, 2023-03-10:
hi @TyshkevichSergey , can u explain more why not using notify characteristic? 
in my case, the service have 4 characteristic like below:
```
 LOG  properties {"Read": "Read", "Write": "Write"}
 LOG  properties {"Read": "Read"}
 LOG  properties {"Read": "Read"}
 LOG  properties {"Notify": "Notify"}
```

Only Notify return value on startNotification