innoveit/react-native-ble-manager #786: BleManagerDisconnectPeripheral event fires twice when peripheral disconnects unexpectedly

soltesza, 2021-10-14:
**Describe the bug**
On Android, when the peripheral is powered off, the BleManagerDisconnectPeripheral event is fired twice.

**To Reproduce**
1. Connect to peripheral
2. Power off peripheral
3. Observe number of BleManagerDisconnectPeripheral events sent to React Native

**Expected behavior**
A single BleManagerDisconnectPeripheral event should be issued

**Smartphone (please complete the following information):**
 - Device: Nexus 6
 - OS: 7.1.1
 - react-native-ble-manager version: 7.6.0
 - react-native version: 0.64.1

**Additional context**
I'm pretty sure this can be solved by just removing the sendConnectionEvent after calling this.disconnect in onConnectionStateChange() in Peripheral.java

mvmmarcus, 2021-11-28:
@marcosinigaglia , i was tested the @soltesza solution and this worked! Can you adjust this?

mvmmarcus, 2021-11-28:
@marcosinigaglia I created this PR with the solution: https://github.com/innoveit/react-native-ble-manager/pull/799

closed 2021-11-29 by marcosinigaglia