innoveit/react-native-ble-manager #791: Unable to send byte array data with write function

Pifoupif, 2021-11-10:
I'm desperately trying to write data to a BLE device and then read its response but I'm unable to. 

I managed to `BleManager.connect` to the required device (`BleManager.isPeripheralConnected` returns `true`). Once I'm connected I `BleManager.retrieveServices` and after I `BleManager.write` passing the correct service and characteristics I get from the device manufacturer (and that I can also log when I `retrieveServices`).

I've tried to `.write` in a `.then` following the `retrieveServices` but also outside of the promise (when pressing a second button in my app).

I need to send this array (I also have other write calls that I tried and it's the same behavior): 
`const arrayData = [1, 3, 1, 11, 0, 1, 244, 52]`

Looking at the documentation the fourth argument of `BleManager.write` has to be a `Byte array`.

So I'm doing this:
`const data = new Uint8Array(arrayData)`
(I've also tried with Buffer: `Buffer.from(data)` and it's the same result)
but I get the following error:

`TypeError: Cannot redefine property: 0
    at Function.defineProperty (<anonymous>)
    at deepFreezeAndThrowOnMutationInDev (deepFreezeAndThrowOnMutationInDev.js:51)
    at deepFreezeAndThrowOnMutationInDev (deepFreezeAndThrowOnMutationInDev.js:66)
    at MessageQueue.enqueueNativeCall (MessageQueue.js:311)
    at Object.nonPromiseMethodWrapper [as write] (NativeModules.js:149)
    at BleManager.js:72
    at tryCallTwo (core.js:45)
    at doResolve (core.js:200)
    at new Promise (core.js:66)
    at BleManager.write (BleManager.js:71)`

I've also tried to pass `arrayData` (as a normal array of int). But in this case my BLE device returns `undefined` (as it expects a byte Array I guess).

Any help would be much appreciated.

"react-native": "0.66",
"react-native-ble-manager": "7.5.0",

Pifoupif, 2021-11-11:
@marcosinigaglia could it be because I have to call `startNotification` before the `write` function as you do in the app example ?

Pifoupif, 2021-11-12:
I found the solution. I needed to call `startNotification` before `write`. My BLE device then sends its response in `handleUpdateValueForCharacteristic`.

closed 2021-11-12 by Pifoupif

AshlandWest, 2023-02-10:
@Pifoupif I'm currently stuck on the same problem, except the characteristic I'm using doesn't support notifications. How did you format the data when you finally got it working?
My UinitArray is logging the message you described in the console and throwing a Malformed Calls from JS error in the device.

ethanstaceyNewr, 2023-09-29:
I have the same issue