innoveit/react-native-ble-manager #793: Error could not find service UUID on peripheral, but retrieveServices there are service in that

canh-eoh, 2021-11-13:
**Describe the bug**
I start notification error "Could not find service with UUID 00001809-0000-1000-8000-00805f9b34fb on peripheral with UUID 9DEF697C-6739-4C59-2313-5476F9E14264", but I check Peripheral info in retrieveServices there are services in that.
**To Reproduce**
Steps to reproduce the behavior:
1. I  build the app on iPhone 7+
2. I scan and connect device BleManager.connect success
3. I BleManager.retrieveServices success
[image]

4. I BleManager.startNotification error
Could not find service with UUID 00001809-0000-1000-8000-00805f9b34fb on peripheral with UUID 9DEF697C-6739-4C59-2313-5476F9E14264"

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: [iPhone7+]
 - OS: [iOS15]
 - react-native-ble-manager version:  "^7.6.3"
 - react-native version: "0.66.1"

joaovitorqueiroz, 2021-11-22:
Hi @canh-eoh I had this same problem, apparently the IOS does not use the full UUID as on android, try to search for the service only with the value "1809".

You can create a constant is to use the Platform to toggle between IOS and Android values.

`const SERVICE =  Platform.OS === 'android' ? '00001809-0000-1000-8000-00805f9b34fb' : '1809';`

canh-eoh, 2021-11-24:
@joaovitorqueiroz thanks it works

closed 2021-11-24 by canh-eoh