innoveit/react-native-ble-manager #795: BleManager.disconnect not working on Android

Dustin-PromenadeSoftware, 2021-11-19:
**Describe the bug**
After calling `BleManager.disconnect(<peripheral id>)` device does not get disconnected even though `BleManagerDisconnectPeripheral` event triggers. 
Checking `BleManager.getConnectedPeripherals([NORDIC_UART_SERVICE_UUID])` still displays the device. 

Only happens on Android. Seems to disconnect fine on iOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Call `BleManager.disconnect(<peripheral id>)`
2. See that `BleManagerDisconnectPeripheral` event triggers
3. Call `BleManager.getConnectedPeripherals()`
4. See peripheral still in list of connected peripherals
5. See on peripheral that it is still connected to phone

**Expected behavior**
Disconnection happens when calling `BleManager.disconnect(<peripheral-id>)`

**Smartphone (please complete the following information):**
 - Device:  Samsung Galaxy S21
 - OS: Android 12
 - react-native-ble-manager version: 8.0.0
 - react-native version: 0.64.1

eastpiger, 2021-11-29:
Same issue here.

Dustin-PromenadeSoftware, 2021-12-09:
@marcosinigaglia 
Any idea of what the cause of this could be? 
Library works fine on iOS but not on Android.

Dustin-PromenadeSoftware, 2021-12-09:
Seems like issue is happening due to startNotification. Running stopNotification does not fix it though. Any ideas?

marcosinigaglia, 2021-12-10:
@Dustin-PromenadeSoftware , the disconnect call is after other operations? Can be also a BLE device problem.

Dustin-PromenadeSoftware, 2021-12-10:
I was able to disconnect just fine if I didn't run any other operations. The issue only happens when I use startNotifications and listen to the characteristic.

knightndgale, 2022-02-10:
still not working, even the example app.js doesn't even disconnect. any fixes?

weimeizhuyi, 2022-04-22:
Same issue here.

weimeizhuyi, 2022-04-22:
> @Dustin-PromenadeSoftware , the disconnect call is after other operations? Can be also a BLE device problem.
This problem can occur on multiple mobile phones

Burzo, 2022-05-05:
Hopefully, this helps someone - in my situation the `BleManagerDisconnectPeripheral` event didn't even trigger upon calling `disconnect`. Basically, nothing happened. Not sure what the root cause is, but I encountered this issue while developing constantly due to hot-reloading (if you terminate and reopen the app it seems the disconnect happens normally behind the scenes).

What helped me out was:

```
if (
    Platform.OS === "android" &&
    (await bleManager.isPeripheralConnected(peripheral.id, [
        SERVICE_READ_UUID,
        SERVICE_WRITE_UUID,
    ]))
) {
    await bleManager.removeBond(peripheral.id)
}
```

Simply run this check when the app is initialized and remove all hanging peripherals.

However, keep in mind that this will completely remove the bond so the device will ask for it again upon reconnection.

closed 2022-06-01 by marcosinigaglia

ManigandanRaamanathan, 2022-12-09:
Same issue, not able to disconnect the ble and also not able to stop notification

digicomarc, 2024-02-28:
For anyone still experiencing a similar issue: This may or may not be the case for you, but after tons of debugging it turned out that my code was connecting to the device multiple times during my connection routine, and calling disconnect would only disconnect one of the multiple connections my app had to the device. To solve this I kept track of what I was already connecting to so I didn't call connect to the same device multiple times. Hope this helps someone!