innoveit/react-native-ble-manager #796: Unable to scan for BLE device

canh-eoh, 2021-11-24:
**Describe the bug**
I BleManager.scan sometimes found, sometimes can't find my BLE device. 
I scan Bluetooth on my phone device, not see, but I try scan on BLE scanner on app nRF Connect is always found.
Do you have any solution for this problem?

**Smartphone (please complete the following information):**
 - Device: [iPhone7+]
 - OS: [iOS15]

mvmmarcus, 2021-11-28:
Hi, @canh-eoh . I was passing something like this, but on Android. The reason is when i was having some error on ble connecting process, the stop scan listener was trigereding. My solution was updated the discoverPeripheral listener when i have some error or when diconnect listener is trigered

szopenkrk, 2022-01-08:
@canh-eoh please show code or try example code from the project. Your issue could be related to access to Bluetooth or the structure of code.

closed 2023-12-05 by marcosinigaglia