innoveit/react-native-ble-manager #797: bleManager.scan not working on Android API 28

deepakjsr7979, 2021-11-28:
I am trying to get available Bluetooth  list of device near by or paired devices ... but scan neither giving any result nor any error.
Only BleManagerStopScan  bleManagerEmitter emiiting after scan done with empty array.

 const removeBond = await bleManager.getDiscoveredPeripherals();
also tried after bleManagerEmitter  called this method. still empty array.

I also tried to directly connect to device using MAC:ID that is also not working.. Please help thanks

phamhien2102, 2021-12-06:
For me, this issue meets on Samsung Ultra 21 5G, Can anyone help me?

Shirsendu-Sarkar, 2021-12-13:
does it have location permission on?

Shirsendu-Sarkar, 2021-12-13:
`@deepakjsr7979 i hope your doing this for scanning peripherals

const delay = milliseconds => {
  return new Promise((resolve, reject) => {
    setTimeout(() => {
      resolve();
    }, milliseconds);
  });
};

await BleManager.scan([], 3, false,{})
await delay(3)
var discoveredPeripherals = await BleManager.getDiscoveredPeripherals([]);
console.log("discoveredPeripherals",discoveredPeripherals)`

phamhien2102, 2021-12-13:
Thank you very much, @Shisendu-Sarkar. I used "react-native-geolocation-service" to turn on location services. Finally, it worked well <3

closed 2021-12-29 by marcosinigaglia