innoveit/react-native-ble-manager #800: Message flooding logcat logs and probably impacting performance. E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation

anwarhamr, 2021-12-05:
This message is showing up every ~100ms and flooding logcat, I can only imagine it's causing and issue w/ performance but in any case it's an error that is annoying and would love to fix it.  As far as I'm aware our BLE device is not and HID device.  
My app connects to 4 BLE devices on an Android OS.  I can reliably send, receive and get notifications to all the services and characteristics we publish.  Has anyone else seen this and know how to fix?

**Smartphone (please complete the following information):**
 - Device: proprietary Mason America tablet
 - OS: Android 8.1
 - react-native-ble-manager version: 7.1.4
 - react-native version: 61.5

marcosinigaglia, 2021-12-14:
Hi @anwarhamr , the library log everything pass through the BLE, maybe is better to add a flag to disable logging data in the writing/reading operations. Or to disable all native loggings.

anwarhamr, 2021-12-14:
@marcosinigaglia is this something that is configured and functionality exists today or something I can change in your the library?

marcosinigaglia, 2021-12-14:
Is something to add in the library

Marvedog, 2022-03-25:
@anwarhamr I'm gonna look into this unless you have already done it. Have you already started with this feature?

anwarhamr, 2022-03-26:
no i have not. I reached out to the vendor that I use for medical Android
devices And they told me to ignore it. that it wouldn't effect performance.
doubtful but I had other things to work on.

iammayankpratapsingh, 2026-10-15:
Hey @marcosinigaglia is this still open? I’d like to pick it up and add the verboseLogging option for Android same as iOS has. Let me know if that works...

marcosinigaglia, 2026-10-16:
> 8\. Hey [@marcosinigaglia](https://github.com/marcosinigaglia) is this still open? I’d like to pick it up and add the verboseLogging option for Android same as iOS has. Let me know if that works...

Hi @marcosinigaglia , I seem to recall that there's only one flag to reduce something. It would be interesting to have different levels: one to disable logging completely, one to leave the logs as is without logging the passing data, and one for full logging.

iammayankpratapsingh, 2026-10-16:
@marcosinigaglia, Sounds good. I’ll add a logLevel option with three levels- none, normal (no data) and verbose (full logs), on both Android and iOS. I’ll keep verboseLogging working too...

closed 2026-10-26 by marcosinigaglia