innoveit/react-native-ble-manager #802: Device data is being split into multiple strings.

victorkvarghese, 2021-12-10:
Hi,

Can some one help please

Eg: ["17 SWIRL 50BBL STAINLESS"] in ios while in android ["17 SWIRL 50BBL STAIN","LESS "]

When issue this library, in iOS i'm able to see data from device. But in android, the same data is split into different strings.
[image]

victorkvarghese, 2021-12-10:
@marcosinigaglia can you please take a look?

marcosinigaglia, 2021-12-14:
Hi @victorkvarghese , I don't understand your problem, are you reading from a characteristic? The MTU in Android is different.

victorkvarghese, 2021-12-14:
@marcosinigaglia I was able to fix it at-last  by setting MTU for android. 
Thanks for the reply and this awesome library :)

closed 2021-12-14 by victorkvarghese