innoveit/react-native-ble-manager #804: How can I get the device thingName with ble-manager?

yoonachen, 2021-12-14:
As title, I want to get thingName from my device.
But, I have no idea about it. 
Anyone know the method to get it?

Thanks!

Shirsendu-Sarkar, 2021-12-28:
const doLoop =async() =>{
      var peripheralId = "id"
      var service = "XXX"
      var characteristic = "YYY"
      var data = "ZZZ"
      // above details must be accurate not dummy and data should be in byte array
      for (let i = 0; i < 2; i++) {
            await BleManager.write(
          peripheralId,
          service,
          characteristic,
          data,
        );
        const response = await BleManager.read(peripheralId, service, characteristic);
        console.log(response)
}

doLoop()

yoonachen, 2021-12-28:
Hey @Shirsendu-Sarkar,  thanks for your response!
But, I want to do like this
```
BleManager.retrieveServices(connectedPeripheral).then((peripheralData) => {
        
  for (i = 0; i < peripheralData.characteristics.length; i++) {

             BleManager.read(
               connectedPeripheral, transparentService, peripheralData.characteristics[i].characteristic
            ).then((data) => { ... }
        }
)}
```
Can I do that?

Shirsendu-Sarkar, 2021-12-28:
@yoonachen are you trying to read every characteristics?

yoonachen, 2021-12-28:
@Shirsendu-Sarkar yes, because my device has wifi module and I want to scan the specific characteristics.
When I tried to use for-loop to read every characteristics, I got read failed. :(

Shirsendu-Sarkar, 2021-12-28:
@yoonachen you need to check property of characteristics before read if it has property read then you could do it else you cant

yoonachen, 2021-12-28:
@Shirsendu-Sarkar yep, I know! I checked it. But it failed when using for loop.

Shirsendu-Sarkar, 2021-12-28:
@yoonachen can you try to console log "peripheralData.characteristics" below for line and print here

yoonachen, 2021-12-28:
@Shirsendu-Sarkar 
This is my peripheralData
```
[{"characteristic": "49535343-1E4D-4BD9-BA61-23C647249616", "isNotifying": false, "properties": ["WriteWithoutResponse", "Notify"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "49535343-8841-43F4-A8D4-ECBE34729BB3", "isNotifying": false, "properties": ["Write"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "49535343-4C8A-39B3-2F49-511CFF073B7E", "isNotifying": false, "properties": ["Write", "Notify"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "77880002-D229-11E4-8689-0002A5D5C51B", "isNotifying": false, "properties": ["Read", "Write", "Notify", "Indicate"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "77880003-D229-11E4-8689-0002A5D5C51B", "isNotifying": false, "properties": ["Read", "Write"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0002-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read", "Write", "Notify", "Indicate"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0003-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0100-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0101-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0102-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0103-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0104-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0105-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0106-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0107-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0108-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C0109-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}, {"characteristic": "FB8C010A-D224-11E4-85A1-0002A5D5C51B", "isNotifying": false, "properties": ["Read"], "service": "49535343-FE7D-4AE5-8FA9-9FAFD205E455"}]

```

Shirsendu-Sarkar, 2021-12-28:
some has read and some does not has read in above json data , so check if property contains read if yes then read it

yoonachen, 2021-12-28:
Hi, I exclude that data which didn't have read property. However, I found that it could not use for loop in retrieveServices.
(My device just has one service called transparent service.)

Shirsendu-Sarkar, 2021-12-28:
BleManager.retrieveServices(id)
      .then(peripheralData => {
        for (i = 0; i < peripheralData.characteristics.length; i++) {
          if (peripheralData.characteristics[i].properties.includes('Read')) {
            BleManager.read(
              id,
              peripheralData.characteristics[i].service,
              peripheralData.characteristics[i].characteristic,
            )
              .then(data => {
                console.log('data', data);
              })
              .catch(err => console.log('read err', err));
          }
        }
      })
      .catch(err => console.log('retreive services err', err));

yoonachen, 2021-12-28:
I’ll try it later.  :)