innoveit/react-native-ble-manager #805: Crashes on Android 12

DanijelBojcic, 2021-12-20:
**Describe the bug**
Android 12 is crashing due to lack of permissions. This happens both in release and debug mode.
Adding the new permissions as described [here](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions) doesn't solves the crash.
I think that a runtime permission check/request is also mandatory.

```
Fatal Exception: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@e67eca18: GattService getDevicesMatchingConnectionStates
       at android.os.Parcel.createExceptionOrNull(Parcel.java:2437)
       at android.os.Parcel.createException(Parcel.java:2421)
       at android.os.Parcel.readException(Parcel.java:2404)
       at android.os.Parcel.readException(Parcel.java:2346)
       at android.bluetooth.IBluetoothGatt$Stub$Proxy.getDevicesMatchingConnectionStates(IBluetoothGatt.java:1842)
       at android.bluetooth.BluetoothManager.getDevicesMatchingConnectionStates(BluetoothManager.java:219)
       at android.bluetooth.BluetoothManager.getConnectedDevices(BluetoothManager.java:179)
       at it.innove.BleManager.getConnectedPeripherals(BleManager.java:37)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:148)
       at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:21)
       at com.facebook.react.bridge.queue.NativeRunnable.run(NativeRunnable.java)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java)
       at android.os.Looper.loopOnce(Looper.java:226)
       at android.os.Looper.loop(Looper.java:313)
       at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:37)
       at java.lang.Thread.run(Thread.java:920)
```

**To Reproduce**
Steps to reproduce the behavior:
1. set `compileSdkVersion` and `targetSdkVersion` to `31`
2. run the app

**Expected behavior**
Not to crash.

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S21 5G
 - OS: Android 12
 - react-native-ble-manager version: 8.0.1
 - react-native version: 0.65.1

marcosinigaglia, 2021-12-20:
Hi, do you use PermissionsAndroid ? (https://reactnative.dev/docs/permissionsandroid)

DanijelBojcic, 2021-12-21:
At the moment of writing the issue I did not realize that those permissions can be requested from react-native.

Should I edit the Readme so that it's compatible with Android 12?

marcosinigaglia, 2021-12-21:
Yes, you can also add the new BLUETOOTH_CONNECT permission required, thanks.

closed 2021-12-21 by marcosinigaglia