innoveit/react-native-ble-manager #807: isPeripheralConnected returns false while I'm connected.

edwardkes, 2021-12-20:
I have a question about the way `isPeripheralConnected` works.
To check if I'm connected to the device, do I have to be connected to it at the App level or at the device level?
I did a test, where I was only connected at the device level to my BT headset and the method returned `false` for the UUID I used.
However, once I also connected to the headset at the App level, I got `true` for the connection for the same UUID.
Is my assumption correct that I need to be connected to the device with the app to see the connection as true?

Silvesterrr, 2022-02-10:
Do only I not understand the question?

coletz, 2022-06-27:
> Do only I not understand the question?

He basically connected to his headphones through his phone's bluetooth settings, then tried to check from his application (using ble-manager lib) if his headphones were reported as connected (trough isPeripheralConnected method, passing the UUID of his headphones). Method returned false.
Then he tried to call the connect method of the library against his headphones and tried to check again the connection status, and this time the method isPeripheralConnected returned true.

So the question is: is connection state returned from isPeripheralConnected method handled at OS level or at application level? And for what I've seen (and what you' ve tried) everything about this method (isPeripheralConnected) is handled at application level from the library (but I haven't seen internal code to be honest)

samknp, 2023-01-25:
Thanks coletz.
Is there any way to discover OS level?

sfsnaruto, 2023-05-28:
same issue even though i already connect to device it is giving false

closed 2023-12-18 by marcosinigaglia