innoveit/react-native-ble-manager #812: Doesn't write or throw error on Android

MrShakes, 2021-12-31:
**Describe the bug**
I'm writing to an external bluetooth device that I've already paired with, code works well on IOS, on Android it works less than 20% of the time.
I'm using the `write` method, code below
```
console.log("send packet1");
BLEWrite(
      connectedId, SERVICE_COMMUNICATION, charId, packet
    )
      .then(() => {
        console.log("send packet2");
      })
      .catch((e) => {
        console.log("Write failed:", e);
      });
```
On IOS I always see the `send packet2` log, on Android I only see it less than 20% of the time, sometimes it completes, sometimes it doesn't and I never get an error message in `catch` which makes me wonder if it hung somewhere in the library

**Expected behavior**
Should successfully write all the time or throw an error if there's one

**Smartphone**
 - Device: Samsung 10
 - OS: Android 11
 - react-native-ble-manager version: 8.0.1
 - react-native version: 0.64

Shirsendu-Sarkar, 2022-01-08:
can you third party app like nrf connect to check write is happening to that particular characteristic?

MrShakes, 2022-01-08:
> can you third party app like nrf connect to check write is happening to that particular characteristic?

Would nrf connect have to first connect to the external device, if so how would the App then connect to the device, how do I do this please?

Shirsendu-Sarkar, 2022-01-08:
> > can you third party app like nrf connect to check write is happening to that particular characteristic?
> 
> Would nrf connect have to first connect to the external device, if so how would the App then connect to the device, how do I do this please?

install  nrf connect app in phone and scan and connect to external device and try to find SERVICE_COMMUNICATION(listed above) then charId(listed above) then try to write and check whether write is happening or not

MrShakes, 2022-01-09:
@Shirsendu-Sarkar thanks however I already mentioned the write works very few times on Android, works all the time(using the same code on IOS) and now adding that it worked using another Android project

KornobisD, 2022-02-13:
Hey, any updates on this issue? Having the same case, on iOS everything works just fine, however on android everything related to .write() function either doesn't work or works once every ~20 times.

MrShakes, 2022-02-14:
@KornobisD unfortunately had to downgrade to the latest 7.x version

PrestonFocusV, 2022-03-01:
> @KornobisD unfortunately had to downgrade to the latest 7.x version

Even on 7.x it seems very hit or miss on Android. 
Does anybody know of a fix?

closed 2023-12-06 by marcosinigaglia