innoveit/react-native-ble-manager #816: writeWithoutResponse no longer resolves promise.

Loxrie, 2022-01-10:
**Describe the bug**

Moving from v7.1.0 to v8.0.1 this code no longer functions:

```
const result = await BleManager.writeWithoutResponse(
          peripheralId,
          SERVICE_UUID,
          WRITE_UUID,
          byteArray,
        );
```

Which causes issues, especially as I need to delay writes to the bluetooth device in question due to it being very slow.

According to BleManager.js, index.d.ts, and Java code it should resolve a promise however the code never does so.  Comparing the Peripheral.java write method between the two versions it's easy to see why:

v7.1.0: [Peripheral.java:716-720](https://github.com/innoveit/react-native-ble-manager/blob/f2fc0861ecd79ef2c7158bf54091bb273304fe79/android/src/main/java/it/innove/Peripheral.java#L716)

```
else if (doWrite(characteristic, data)) {
    Log.d(BleManager.LOG_TAG, "Write completed");
    if (BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE == writeType) {
        callback.invoke();
    }
```

v8.0.1: [Peripheral.java:852-857](https://github.com/innoveit/react-native-ble-manager/blob/4cdc595cb3e4cbcb486c03077c1c835b2b697b3c/android/src/main/java/it/innove/Peripheral.java#L852)

```
		if (data.length <= maxByteSize) {
			if (! doWrite(characteristic, data, callback)) {
				callback.invoke("Write failed");
				writeCallback = null;
			}
		} else {
```

however if the data is > maxByteSize this code would execute the callback, line 902.

**To Reproduce**
Steps to reproduce the behavior:
1. Call writeWithoutResponse
2. Await or then the promise, nothing will happen.

**Expected behavior**

The promise should be resolved if the task is successful.

**Screenshots**

N/A

**Smartphone (please complete the following information):**

 - Device: Samsung A3
 - OS: 8.0.0
 - react-native-ble-manager version: v8.0.1
 - react-native version: 0.66.4

**Additional context**

This issue does not occur on iOS, I think the writeWithoutResponse method in BleManager.m is resolving the callback.   Shout if you'd like a PR, for now I've simply changed it to read this:

```
if (data.length <= maxByteSize) {
			if (! doWrite(characteristic, data, callback)) {
				callback.invoke("Write failed");
				writeCallback = null;
			} else {
				callback.invoke();
			}
		} 
```

SCasarotto, 2022-01-12:
We also saw issues in v8 around writing on Android. We reverted back to 7.6.3 in order to deploy.

Loxrie, 2022-02-03:
Yes, I can do this early next week.

Loxrie, 2022-02-03:
Sorry the latest release is the one I've raised the error against.  Is there a newer version than 8.0.1? @marcosinigaglia

marcosinigaglia, 2022-02-03:
The latest is 8.0.2

Loxrie, 2022-02-03:
Ah OK.  It's not showing for me as a release/tag, I'll check it out though.

xand3r40r93, 2022-02-08:
I have similar trouble with version 8.0.2.

PrestonFocusV, 2022-03-01:
> I have similar trouble with version 8.0.2.

I am also on 8.0.2 & face the same issue

PrestonFocusV, 2022-04-04:
@marcosinigaglia The latest version is not being fetched when installed via _'npm install react-native-ble-manager'_. 

I had to clone the repo & set npm to install from there, but then I was able to confirm this issue has been resolved.

closed 2022-06-06 by marcosinigaglia