innoveit/react-native-ble-manager #818: How to check if device is already connected to another bluetooth device from default settings on phone

frankied003, 2022-01-13:
I'm attempting to add a function within my application that checks if any bluetooth devices are already connected to the phone. These connections happen outside the app and in the bluetooth settings of the phone. 

One issue that has happened is that this device that we are checking for is not coming up as a device while using the example code, but coming up in the default app settings. 

The device is a [Parani](http://www.senanetworks.com/sd1000-01.html) by the way.

KPS250, 2022-01-28:
You can use isPeripheralConnected whenever app comes in foreground. 

```
# isPeripheralConnected(peripheralId, serviceUUIDs)
# Check whether a specific peripheral is connected and return true or false. Returns a Promise object.

#Example:

BleManager.isPeripheralConnected(
  "XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX",
  []
).then((isConnected) => {
  if (isConnected) {
    console.log("Peripheral is connected!");
  } else {
    console.log("Peripheral is NOT connected!");
  }
});
```

zacdemi, 2022-04-04:
What if you don't have the UUID? Is there a way to get a list of device names and connection status from settings?