innoveit/react-native-ble-manager #820: iOS Connect Error: Peer removed pairing information

KPS250, 2022-01-28:
Scenario:
1) User paired ble device with passkey on iOS device.
2) The passkey for ble device changed now
3) User tries to reconnect but gets below error
<CBPeripheral: 0x282385360, identifier = D15AC6F9-6A69-B72A-8EC8-BC6382AC2CD8, name = 434992, mtu = 185, state = disconnected>. (Peer removed pairing information)

Here theres no way to programatically remove the pairing after which we can start with fresh pairing mechanism. User needs to manually goto bluetooth settings and do forget device which is not a good user experience. Also my ble device passwords are resets frequently which makes this manual resetting frustrating.

pgpankajgoyal, 2022-02-22:
I have same issue, any solution for this?

Koxx3, 2022-06-09:
same issue for me too

lauritaipale, 2022-06-16:
It's an iOS feature/limitation, individual apps can't remove Bluetooth device pairings. The best you can do in your app is to instruct/redirect the user to remove the existing pairing from the device's Bluetooth settings when you detect that particular error.

closed 2022-06-16 by marcosinigaglia