innoveit/react-native-ble-manager #822: Cannot read properties of undefined (reading 'start')]

mikasmart, 2022-02-01:
**Describe the bug**
Can't start BleManager on Android

**To Reproduce**
Steps to reproduce the behavior:
Create a react native app
Add react-native-ble-manager@8.0.1
Create a little script in App.js

`
import React, { useEffect } from 'react';
import { Text, View } from 'react-native';
import BleManager from 'react-native-ble-manager';

const App = () => {
	useEffect(() => {
		BleManager.start({ showAlert: false });
	}, []);

	return (
		<View style={{ flex: 1, justifyContent: 'center', alignItems: 'center' }}>
			<Text>Try editing me! 🎉</Text>
		</View>
	);
};

export default App;
`

**Expected behavior**
[Unhandled promise rejection: TypeError: Cannot read properties of undefined (reading 'start')]
at node_modules\react-native-ble-manager\BleManager.js:236:17 in Promise$argument_0
at node_modules\react-native\node_modules\promise\setimmediate\core.js:45:4 in tryCallTwo
at node_modules\react-native\node_modules\promise\setimmediate\core.js:200:12 in doResolve
at node_modules\react-native\node_modules\promise\setimmediate\core.js:66:2 in Promise
at node_modules\react-native-ble-manager\BleManager.js:232:11 in start
at App.js:7:13 in useEffect$argument_0
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:19280:19 in invokePassiveEffectCreate
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:93:9 in invokeGuardedCallbackProd

**Smartphone (please complete the following information):**
 - Device: RMX2001 (RealMe 6)
 - OS: Android 11
 - react-native-ble-manager version: 8.0.1
 - react-native version: 0.64.3

marcosinigaglia, 2022-02-02:
The library is not installed correctly.

closed 2022-02-02 by marcosinigaglia

oliuradu, 2022-06-11:
> 

how we have to install it?

He added the library in the package.json
Added the permissions needed, what else does he need to do?

jamesmacwilliam, 2023-02-22:
@mikasmart did you end up resolving this?  I'm seeing something similar

mikasmart, 2023-03-03:
Yep, i change for VueJS + Capacitor and its work better !

https://capacitorjs.com/solution/vue
https://github.com/capacitor-community/bluetooth-le

It was by realizing this POC that I understood that React Native was not suitable for bluetooth BLE. Maybe since then it has improved.