innoveit/react-native-ble-manager #830: What does it mean about `Connection error` ?

eastpiger, 2022-02-08:
I just upgrade ble-manager to the newest version. After upgrading, I find connect() may throw `Connection error` sometimes, and every thing goes well after retry connecting. 

This error is meaningless and What does it mean?

madfauzi, 2022-02-09:
same issue.

marcosinigaglia, 2022-02-09:
@eastpiger check the device log.

pankajdreamsoft, 2022-02-09:
Same issue.

lucasahlgren, 2022-03-31:
@pankajdreamsoft Did you find a solution? I am facing the same problem on Android.

pgpankajgoyal, 2022-04-01:
@lucasahlgren : Yes, I did change my native code to get proper response. Which type of device you are trying to connect ?

lucasahlgren, 2022-04-02:
I managed to sort it out with another package. Thank you anyways :)

lucasahlgren, 2022-04-02:
@pgpankajgoyal I managed to sort it out with another package. Thank you anyways :)

LazloN5, 2022-04-26:
@lucasahlgren what was the package you used?

closed 2022-06-17 by marcosinigaglia