innoveit/react-native-ble-manager #831: Stuck on retrieveServices (not connected after connection?)

Silvesterrr, 2022-02-10:
**Describtion**
Hi, I've got this annoying bug it appears inconsistently every few times. It's really similar to this Issue  [#402]
(https://github.com/innoveit/react-native-ble-manager/issues/402)
It looks something like this:
 * Connect do device
 * Retrieve services -> and at this point the code hangs up
I never found out this bug after first run of my app (first connect). Only at  second, etc. connection.

I can't find any mistakes because one time it works like a charm and other time it has a problem.
I found out that disabling and enabling Bluetooth on the phone and retrying connecting works
(But I can't make my users to disable and enable Bluetooth. Or can I ;) ha-ha).

**My code**
I've tried writing my code in 2 ways both have this bug.
```
async function internalConnect() {
    await BleManager.connect(item.id);
    console.log('after conn');
    await BleManager.retrieveServices(item.id);
    console.log('after retrive');  // <---if the bug appears I never get to see this log
    await BleManager.write(
      item.id,
      global.GATTS_SECURITY_UUID,
      global.GATTC_ACCESS_CODE_UUID,
      stringToIntArray(code),
    );
    console.log('after write');
    await BleManager.read(
      item.id,
      global.GATTS_SECURITY_UUID,
      global.GATTC_ACCESS_CODE_UUID,
    )
      .then(readData => {
        console.log('access code response: ' + readData);
       //My other stuff here
      })
      .catch(error => {
        // Failure code
        console.log(error);
      });
  }
```
And this way:
```
BleManager.connect(item.id)
      .then(() => {
        console.log('connected to device');
        setTimeout(() => {
          BleManager.isPeripheralConnected(item.id, []).then(isConnected => {
            //do not remove
            if (isConnected) {
              console.log('Peripheral is connected!');
            } else {
              console.log('Peripheral is NOT connected!');
              Connect(); //<--- run the same function
              return;
            }
          });
        }, 900);
        setTimeout(() => {
          console.log('timeout');

          BleManager.retrieveServices(item.id)
            .then(PeripheralInfo => {
              console.log(PeripheralInfo);

              BleManager.write(
                item.id,
                global.GATTS_SECURITY_UUID,
                global.GATTC_ACCESS_CODE_UUID,
                stringToIntArray(code),
              )
                .then(() => {
                  console.log('written');
                  BleManager.read(
                    item.id,
                    global.GATTS_SECURITY_UUID,
                    global.GATTC_ACCESS_CODE_UUID,
                  )
                    .then(readData => {
                      console.log('access code response: ' + readData);
                      //My stuff here
                    })
                    .catch(error => {
                      handleConnectionErrors(error);
                      return;
                    });
                })
                .catch(error => {
                  handleConnectionErrors(error);
                  return;
                });
            })
            .catch(error => {
              handleConnectionErrors(error);
              return;
            });
        }, 900);
      })
      .catch(error => {
        handleConnectionErrors(error);
        return;
      });
```
In this way when the bug appears after running ```isPeripheralConnected``` I get ```Peripheral is NOT connected!``` So I've tried to resolve this by running this function again, but it just loops in this place and doesn't work.

**How to Reproduce**
Accually I don't know. It appears so random, I can't control it. I only know that it appears at second or next connections.

**Smartphone:**
 - Device: xiaomi redmi note 7
 - OS: android 10 QKQ1.190910.002
 - react-native-cli: 2.0.1
 - react-native: 0.66.4
 - ble-manager: 8.0.1s

Btw, I really like this library. It's really intuitive :)

liuliu66666, 2022-02-11:
same issue

michael-jogo, 2022-03-21:
@Silvesterrr any update on this?

Silvesterrr, 2022-03-21:
> @Silvesterrr any update on this?

Nope, am focused on something different...
But I'll be back onto it soon. 
If I figure out anything new I'll post it.

sarevok89, 2022-09-29:
Is there any update on this issue? I'm currently struggling with the same issue.

sundayhd, 2022-11-19:
Same problem :-(
as workaround im wating at least 1 sec and if there is no connection I trigger disconnect and then reconnect again every time. something like this:

`const connectToDevice = async (id) => {
    try {
      await BleManager.connect(id);
    }
    catch(error) {
      handleError(error);
      return connectToDevice(id);
    }
    Log("Weight", "setDevice", "Connected to:", id);
    clearTimeouts();
    setConnectionStatus(1);
    currentDeviceId.current = id;
    retrieveServices(id);
    handleHangOutBug(id);
  }

  const handleHangOutBug = (id) => {
    hangOutTimeout.current = setTimeout(() => {
      if (connectionStatus.current === 1) {
        hangOutretries.current = hangOutretries.current + 1;
        isConnectingProcess.current = true;
        disconnect().then(() => {
          connectToDevice(id);
        });
      }
    }, (1000 + (hangOutretries.current * 250)));
  }`

sundayhd, 2022-11-22:
Toady I play with "bluetooth MAP version & AVRCP version" inside developer options.
When I changed map version to 1.3 its looks like that problem go away, so give it a shot 😃

Silvesterrr, 2022-12-14:
> Toady I played with "bluetooth MAP version & AVRCP version" inside developer options. When I changed map version to 1.3 its looks like that problem go away, so give it a shot 😃

yea, fun fact but I cannot force user to change that. Anyone tried using "react-native-ble-plx"?
Does this library has this issue?

sundayhd, 2022-12-27:
> yea, fun fact but I cannot force user to change that. Anyone tried using "react-native-ble-plx"?
> Does this library has this issue?

I already tried ble-plx, same issue 😟

Silvesterrr, 2023-01-11:
> Did you found a solution for this issue ?
Sorry I did not. I kinda hacked the way throught by disabling bluetooth and enabling it when the problem occures with other library. But I hope this issue will be resolved because it's so frustrating.

JoshuaGuzman02, 2023-02-06:
Mismo problema, alguna solución?

sfsnaruto, 2023-05-28:
Same issue

KonstantinKostianytsia, 2023-05-28:
The same issue

maeda-kazuya, 2023-07-05:
Same here, any solutions for this?

sfsnaruto, 2023-07-05:
@maeda-kazuya , @KonstantinKostianytsia  try to run the scan method first and then call the connect method.

maeda-kazuya, 2023-07-06:
Thanks, I just came up with some workaround.
I added retry process as follows and it works for now.
(Still wondering the root cause of failure..)

```
    let isConnected = false;

    while (!isConnected) {
      await BleManager.connect(id)
        .then(() => {
          isConnected = true;
        })
    }
```

cfradella, 2023-09-06:
Also encountering this issue. It'll stall on `retrieveServices` until something else interrupts it.

froggydisk, 2023-09-14:
I don't know if this could be an answer, but let me share my code.

Before, you should see two parts of the original code.
```javascript
// react-native-ble-manager > src > types.ts
export interface PeripheralInfo extends Peripheral {
  serviceUUIDs?: string[];
  characteristics?: Characteristic[];
  services?: Service[];
}
```
```javascript
// react-native-ble-manager > src > index.ts
retrieveServices(peripheralId: string, serviceUUIDs: string[] = []) {
    return new Promise<PeripheralInfo>((fulfill, reject) => {
      bleManager.retrieveServices(
        peripheralId,
        serviceUUIDs,
        (error: string | null, peripheral: PeripheralInfo) => {
          if (error) {
            reject(error);
          } else {
            fulfill(peripheral);
          }
        }
      );
    });
  }
```
When you use retrieveServices, you should return the result 'peripheral', whose elements' types are defined by Interface 'PeripheralInfo'. As you can see, the original code makes PeripheralInfo extend Peripheral, and Interface 'Peripheral' is as below.
```javascript
export interface Peripheral {
  id: string;
  rssi: number;
  name?: string;
  advertising: AdvertisingData;
}
```

These say, the result of retrieveServices must have attributes 'id', 'rssi', and 'advertising' while others are optional.
BUT, what if the result does NOT have the attributes?
You'd better not force it to distribute the information of what it doesn't have.

So, this is the part of my code. 
```javascript
export interface PeripheralInfo { **-> this is what I fixed**
  id: string; **-> this is what I added**
  serviceUUIDs?: string[];
  characteristics?: Characteristic[];
  services?: Service[];
}
```

In my code, PeripheralInfo does not extend Peripheral anymore. Instead, the result must contain 'id', because you offered it when you call retrieveServices function. 
You can put other options if you want, like this.
```javascript
rssi?: number;
name?: string;
advertising?: AdvertisingData;
```
Don't forget to make them optional by using '?'

2sem, 2023-09-15:
sometimes connecting is fast, but retrieveServices takes over 30 seconds.
And then disconnecting also takes much long time.

RakeebBaig1, 2023-10-11:
react-native-ble-plx is better than react-native-ble-manger.
its enables us to retireveServices on both platform Android & IOS and perform any thing you can .

MikaelErikSvensson, 2023-10-23:
My current workaround is something like this, where I basically race it against another promise. If the other promise resolves first I consider it a timeout and handle it accordingly.

```sh
const tryGetServices = await Promise.race([
	BleManager.retrieveServices(peripheral.id),
	new Promise((resolve, reject) =>
		setTimeout(() => {
			resolve('timeout')
		}, 10000)
	),
])

const peripheralData = tryGetServices as PeripheralInfo | string

if (typeof peripheralData === 'string') {
	// timeout
} else {
	// got services, proceed...
}
```

Loovery, 2023-12-24:
Any update? I have same problem after upgrade version 8.x.x to 11.x.x

Loovery, 2023-12-26:
What I found out.
1) The problem does not occur on all devices. 100% of the problem is reproduced with older Wahoo kickr (CyclingPower Service).
2) The device connects without problems. All data is read. The problem is exactly when retrieveServices is requested. It just hangs without any response.

The function **retrieveServices**, reaches the code: 
```
if !uuids.isEmpty {
                peripheral.instance.discoverServices(uuids)
            } else {
                peripheral.instance.discoverServices(nil)
            }
```
Nothing further

Loovery, 2023-12-27:
Version 11.0.8 fixed my problem

vhakulinen, 2024-04-25:
I just ran across this issue with a peripheral that is already "connected" (i.e. `BleManager.isPeripheralConnected` returns true).

It looks like that the command queue is the one that drops the ball. If the `gatt` is null, then it just bails out and doesn't invoke any pending callbacks with errors.

dejawho, 2024-05-13:
same problem, if the device is really connect retrieveService never return

lzagar-itm, 2024-05-27:
Running version `11.4.0` and still having issue. It hangs on `retrieveServices`, I tried to refresh cash, disconnect before I connect, put timeout before calling `retrieveServices` but still no luck. I looked at native code and `onServicesDiscovered` never gets called. Strange thing is that I can connect first time to it but after that I have to remove and add device again. I got Google Pixel 8 running Android 14. If creator someone has any input why this might be happening or where to start to look keep me posted

keyre, 2024-07-12:
> My current workaround is something like this, where I basically race it against another promise. If the other promise resolves first I consider it a timeout and handle it accordingly.
> 
> ```shell
> const tryGetServices = await Promise.race([
> 	BleManager.retrieveServices(peripheral.id),
> 	new Promise((resolve, reject) =>
> 		setTimeout(() => {
> 			resolve('timeout')
> 		}, 10000)
> 	),
> ])
> 
> const peripheralData = tryGetServices as PeripheralInfo | string
> 
> if (typeof peripheralData === 'string') {
> 	// timeout
> } else {
> 	// got services, proceed...
> }
> ```

It works on  android!

My solution is:

```js

 myRetrieveServices = async () => {
    return await Promise.race([
      BleManager.retrieveServices(this.deviceConnected.id),
      new Promise((resolve, reject) => setTimeout(reject, 3000)),
    ]);
  };

```

vitorsilvalima, 2025-05-21:
I'm also having the same issue described in this issue. Some users from time to time get this error where trying to retrieve services from the printer hangs forever

therealkeeg1, 2026-03-04:
I fixed this issue on iOS with my recent PR.

closed 2026-10-26 by marcosinigaglia