innoveit/react-native-ble-manager #834: startNotificationUseBuffer - buffer not working

alxpub, 2022-02-21:
**Describe the bug**
startNotificationUseBuffer - not buffering indications

**To Reproduce**
use startNotificationUseBuffer

**Expected behavior**
i expect that startNotificationUseBuffer is return array of buffered values

**Smartphone (please complete the following information):**
 - any android device or emulator
 - OS: android 11
 - react-native-ble-manager version: 8.0.1
 - react-native version: 0.64.3

**Additional context**
I think it removed here
https://github.com/innoveit/react-native-ble-manager/commit/7cf4da77b9bec054984cb60924c28f08194afd29#diff-c9831a4c6df02b6e48853d646505e62b282b35fa028ddaaee1ed5092ea48e5afL513

rakeshostwal1, 2022-10-16:
startNotificationWithUseBuffer is still breaking for me, the read notification callback returns [0,0,0,0] Byte Array when BLE sends the Message to mobile app.

swey, 2022-11-14:
> startNotificationUseBuffer is still breaking for me

Yes, same here. After upgrading to the latest version of `react-native-ble-manager`, I receive wrong values when using `startNotificationUseBuffer`.

swey, 2022-11-17:
@marcosinigaglia Please re-open the issue 🙏

kscgl, 2023-03-11:
I am having the same issue. `startNotification` returns expected values but `startNotificationUseBuffer` returns only `0`'s.

schimini, 2023-04-04:
It says here that the #866 fixed it but actually it changed behavior. I think the docs should change.

marcosinigaglia, 2023-04-05:
Hi @schimini , can you please make a PR with the doc changes?

schimini, 2023-04-05:
@marcosinigaglia ofc! Btw would it be cool if I also implemented some sort of throttling behavior under some other method, The idea to reduce react native bridge calls is a good one but with a bytes buffer approach I feel that the handling is more granular than what the current users would like to use it as. I will do one separate PR for this sometime today If you think it makes sense.

marcosinigaglia, 2023-04-05:
@schimini we can talk about it in a separate issue or discussion.

schimini, 2023-04-05:
@marcosinigaglia the pr related docs is done. having gone through the code I think I found a bug so I created another one.

schimini, 2023-04-12:
@kscgl @rakeshostwal1 @swey would you mind performing tests on this in when #1003 is done?

nelson-saldanha, 2023-05-04:
Maybe I'm misreading but shouldn't this line on NotifyBufferContainer.java:22
`System.arraycopy(value, toInsert.length, rest, restLength, rest.length);`
be like this:
`System.arraycopy(value, toInsert.length, rest, 0, rest.length);`
So we preserve the remaining data?

schimini, 2023-05-05:
@nelson-saldanha you're right! can you make the fix for this? (i don´t know how i didn't see it)