innoveit/react-native-ble-manager #838: Bluetooth Peripheral Stuck in "Connecting" State On iOS

pgpankajgoyal, 2022-03-02:
**Describe the bug**
I am trying to connect ble A&D device but stuck on "Connecting to peripheral with UUID : xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx".

It works with my NONIN (Oximeter) but fail when trying to connect with A&D (weight and BP) machine on iOS device.

It's perfect working for android devices.

**To Reproduce**
Steps to reproduce the behavior:
1. Run this sample application and enable bluetooth on iOS device.
2. Take weight on A&D device but fail to connect.

**Expected behavior**
Want to connect A&D machine (Weight and BP) with iPhone.

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone 11]
 - OS: [e.g. iOS 14.8.1]
 - react-native-ble-manager version: 8.0.1
 - react-native version: 0.63.4

marcosinigaglia, 2022-03-03:
Hi, this is not a issue, is how it works in iOS.

closed 2022-03-03 by marcosinigaglia

priyasamyal, 2024-01-02:
Hi @pgpankajgoyal  Any update on this. I am also facing the same issue for A&D BP device in iOS