innoveit/react-native-ble-manager #848: Crash on android 12

ndebartha, 2022-04-12:
Getting unexpected crash on android 12. Anything do we need to add/check?

java.lang.SecurityException: 
  at android.os.Parcel.createExceptionOrNull (Parcel.java:2437)
  at android.os.Parcel.createException (Parcel.java:2421)
  at android.os.Parcel.readException (Parcel.java:2404)
  at android.os.Parcel.readException (Parcel.java:2346)
  at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner (IBluetoothGatt.java:1886)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration (BluetoothLeScanner.java:519)
  at android.bluetooth.le.BluetoothLeScanner.startScan (BluetoothLeScanner.java:305)
  at android.bluetooth.le.BluetoothLeScanner.startScan (BluetoothLeScanner.java:161)
  at android.bluetooth.BluetoothAdapter.startLeScan (BluetoothAdapter.java:6057)
  at android.bluetooth.BluetoothAdapter.startLeScan (BluetoothAdapter.java:5959)
  at it.innove.LegacyScanManager.scan (LegacyScanManager.java:59)
  at it.innove.BleManager.scan (BleManager.java:203)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.facebook.react.bridge.JavaMethodWrapper.invoke (JavaMethodWrapper.java:371)
  at com.facebook.react.bridge.JavaModuleWrapper.invoke (JavaModuleWrapper.java:150)
  at com.facebook.react.bridge.queue.NativeRunnable.run (Native Method)
  at android.os.Handler.handleCallback (Handler.java:938)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage (MessageQueueThreadHandler.java:26)
  at android.os.Looper.loopOnce (Looper.java:226)
  at android.os.Looper.loop (Looper.java:313)
  at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run (MessageQueueThreadImpl.java:225)
  at java.lang.Thread.run (Thread.java:920)

Cosmin-Madalin, 2022-04-12:
I think you are missing some runtime permissions: BLUETOOTH_SCAN, BLUETOOTH_CONNECT

Cosmin-Madalin, 2022-04-12:
Yes, but they are runtime permissions. Having them in the manifest does not suffice, you need to explicitly ask the user to grant them.

ndebartha, 2022-04-12:
Yes you are correct. I missed that. Thank you!

closed 2022-04-15 by marcosinigaglia