innoveit/react-native-ble-manager #849: Improve Events (or update Docs)

JThobbyist, 2022-04-16:
Events should be listenable directly on the BleManager instance. The docs show something like 
```
import { NativeModules, NativeEventEmitter } from 'react-native';
BleManagerEventEmitter = new NativeEventEmitter(BleManagerModule);
```

This should be included out-of-the-box by adding the above 'import' statement as well as the following class-edits to BleManage.js:

```
class BleManager extends NativeEventEmitter {
constructor(){
super(BleManagerModule);
...
}
...
}```

And if this is already the case that BleManager exposes events itself, then the docs should be updated (but it doesn;t look like it does based on inspection of the code)

marcosinigaglia, 2022-07-25:
Hi, I made the example like the official [doc](https://reactnative.dev/docs/native-modules-android#sending-events-to-javascript), why are you suggesting to do it in another way?

JThobbyist, 2022-07-25:
Hi Marco,

I am suggesting to do it another way because the official doc example is a use-case where you can choose if you want the eventemitter or not. This choice works well for use-cases where the eventemitter is not required for the module to function in a useful way. 

The react-native-ble-manager module relies directly on the events, and would not work without them. As it currently is, the module forces every user to repeat the code from the example in the docs in order to use the module.

Therefore, my suggestion is to streamline the user experience by incorporating the example from the docs “under the hood” within the module. This way, users don’t need to repeat it, and can simply listen for events on the module itself. 

Please let me know your thoughts?

Best,
-Joe

closed 2024-11-22 by marcosinigaglia