innoveit/react-native-ble-manager #850: Help Needed To Write Data on BLE Device

jaferi512, 2022-04-21:
Version

react-native-ble-manager: ^8.0.2
react-native: ^0.68.1
iOS/Android: 11
Expected behaviour

LED Light Strip that's connected to the device should change colors on WriteWithoutResponse call.

Actual behaviour

WriteWithoutResponse passes successfully but device shows no change in color.

Steps to reproduce

Successful connection to LED Light Strip
Send WriteWithoutResponse Call
Here is the Response from BLE Device That I'm getting from LED Strips. I want to write on BLE Device and there is only one service that has property to writewithoutresponse but I'm confused a bit that how can I write on BLE By using This Characteristic.
I'm Using Below Function But I'm unable to figure out that which data will be passed:
    BleManager.writeWithoutResponse(
      peridata.id,
      fullUUID(peridata.services[2].uuid),
      fullUUID(peridata.characteristics[4].characteristic),
      [0xFF, 0, 0],
    )
      .then(() => {
        console.log('Success');
      })
      .catch(error => {
        console.log(error);
      });
[image]
l

anees17861, 2022-04-30:
@jaferi512 Try sending data through a ble app like Nrf Connect. This way you will at least come to know if it's led strip issue or wrong data issue. If it works over there, then only try to debug the app side.
One more recommendation would be to use constant values of service and characteristic uuid instead of getting from some array. It may not be guaranteed that the services and characteristics would always be in same order for every connection

jaferi512, 2022-05-17:
@anees17861 Device is Connecting Successfully But when I try to read or write the data from it, it doesn't give response to me. If You've worked on Such Kind Of Project, Please Share some Code Sample To Get Familiar with it.
Thanks Much.

anees17861, 2022-05-17:
I have worked on projects involving Bluetooth, but the main thing is it depends on the device as well. Different manufacturers generally implement their own solutions instead of following a standard so the data frame required depends on the manufacturer.

I see that you are sending [0x00, 0x01, 0x02]. Where did you get this frame from? Is it in the device specs or something random you came up with?

jaferi512, 2022-05-17:
@anees17861 Hardware developer has told me to send values like this.That's why I'm trying like this.

anees17861, 2022-05-17:
@jaferi512 Please try the following things:
1. Make sure service discovery is complete before writing
2. Hardcode service and characteristic to write to instead of picking from array
3. If the hardware developer is in your team, ask him to check logs on their side as well

closed 2023-12-04 by marcosinigaglia