innoveit/react-native-ble-manager #853: Device not finding all BLE server services

laurislopata, 2022-04-28:
**Describe the bug**
After creating a custom service using bluetoothctl scan does not discover it.

**To Reproduce**
Before scanning for devices, follow the first answer on this StackOverflow post: https://stackoverflow.com/questions/25427768/bluez-how-to-set-up-a-gatt-server-from-the-command-line to create a custom Service to check if the phone is discovering them. and when you register the service, one extra step you should do is execute menu advertise and service 0xFFFF in the bluetoothctl shell with this the new service will be advertised as well

And when in the app just press "SCAN BLUETOOTH" and when scanning finishes click on your laptops name in the console it will show Connected to XX:XX:XX:XX:XX:XX and after this click the other button and it should print the UUIDs of all the services. For me it only prints the default 3: 1800, 1801, 180a but not ffff, and show's error: Characteristic 0000aaaa-0000-1000-8000-00805f9b34fb not found.

**Expected behavior**
Discovering all the services

**Smartphone (please complete the following information):**
 - Device: Pixel 6
 - OS: Android 12
 - react-native-ble-manager version: ^8.0.2
 - react-native version: 0.68.1

Thank you in advance for the feedback, if anyone wants to see my code it's here: https://github.com/laurislopata/ble-manager-prototype

marcosinigaglia, 2022-04-28:
Hi @laurislopata , bluetoothctl is used to create a Buetooth LE or Buetooth service?

laurislopata, 2022-04-28:
I used it to create a GATT service for Bluetooth LE