innoveit/react-native-ble-manager #857: Arduino + BLE (HC-05 bluetooth device) + React Native = Device not found

Ankiewicz, 2022-05-30:
**Before open an issue**
- Check the closed issues, your question maybe is not new. [Looked, not there]
- We can't debug or reproduce hardware issue. 
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
I am using this package on a hardware prototype which works just fine. I am setting up another project and I am prototyping out the concept with Arduino components. 

My issue is, my computer and my Android phone both find the Arduino BLE device just fine but my application using this package does not find it at all. I do find other devices just not the one arduino ble device. 

Shouldn't the react application be able to find all of the same devices that my phone does?

**Expected behavior**
I expect all available BLE devices to display

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Galaxy Note S10
 - react-native-ble-manager version:  8.1.0
 - react-native version: 0.64.2

```
import React, {
    useState,
    useEffect,
  } from 'react';
  
  import {
    SafeAreaView,
    StyleSheet,
    Image,
    TextStyle,
    ViewStyle,
    ImageStyle,
    ScrollView,
    View,
    Text,
    StatusBar,
    NativeModules,
    NativeEventEmitter,
    Button,
    Platform,
    PermissionsAndroid,
    FlatList,
    TouchableHighlight,
  } from 'react-native';
  import { observer } from "mobx-react-lite"
  
  /**
   * Sample BLE React Native App
   *
   * @format
   * @flow strict-local
   */
  
  
  import BleManager from 'react-native-ble-manager';
  const BleManagerModule = NativeModules.BleManager;
  const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);
  
  const FULL = {
    flex: 1,
  }
  const HEADER = {
    paddingBottom: 12,
    paddingHorizontal: 12,
    paddingTop: 12,
  }
[... 294 lines cut ...]
  
              <View style={{margin: 10}}>
                <Button title="Retrieve connected peripherals" onPress={() => retrieveConnected() } />
              </View>
  
              {(list.length == 0) &&
                <View style={{flex:1, margin: 20}}>
                  <Text style={{textAlign: 'center'}}>No peripherals</Text>
                </View>
              }
            
            </View>              
          </ScrollView>
          <FlatList
              data={list}
              renderItem={({ item }) => renderItem(item) }
              keyExtractor={item => item.id}
            />              
        </SafeAreaView>
      </View>
    )
  })
  
  export default BLEConnectScreen
```

marcosinigaglia, 2022-06-14:
Hi @Ankiewicz , maybe the device is aready connected to the phone.

Ankiewicz, 2022-06-14:
I cleared it from the Bluetooth history and everything. I was hoping that was the issue but it wasn't.

marcosinigaglia, 2022-06-16:
If you use LightBlule app do you see the device in the list?

closed 2023-03-01 by marcosinigaglia