innoveit/react-native-ble-manager #865: lambda expressions are not supported in -source 7

quangcanh2975, 2022-06-10:
Hi. 
I'm not sure if I'm doing wrong but when running app using command: `npx react-native run-android` or building with Android Studio, it always returns this error:
node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManager.java:747: 
error: lambda expressions are not supported in -source 7
`            scanManager.stopScan(args -> {});`
                                      ^
  (use -source 8 or higher to enable lambda expressions)
**To Reproduce**
Steps to reproduce the behavior:
1. After install all packages
2. Start app by command: npx react-native run-android 
3. See error

**Expected behavior**
It must build the app successfully.

**Screenshots**
Having the same error when building by Android Studio:
[image]

**Smartphone (please complete the following information):**
 - Device: Samsung A52s 5G
 - OS: Android 12
 - react-native-ble-manager version: 8.3.0
 - react-native version: 0.66.0

**android/build.gradle:
```
buildscript {
    ext {
        buildToolsVersion = "30.0.2"
        minSdkVersion = 23
        compileSdkVersion = 31
        targetSdkVersion = 31
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath('com.android.tools.build:gradle:3.6.0')
    }
}
```

FroxCode, 2022-06-14:
I am getting the same error using with this library. 

It worked up until recently, I am not sure what changed.

marcosinigaglia, 2022-06-14:
Hi, you are using and old java version to compile.

closed 2022-06-14 by marcosinigaglia

dmingjuan, 2022-06-22:
+1. 
This problem does not exist in version 7.6，but exists in version 8.4.
I have configured :
android {
    .....
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
}

dmingjuan, 2022-06-22:
> 

> you are using and old java version to compile.

I have used the java8 version, but this problem still occurs

jtovarto, 2022-06-27:
> compileOptions
Hi, where did you configured this options?

jtovarto, 2022-07-04:
> Hi, you are using and old java version to compile.

I have the same problem, I see that you are referring to an old version of java, but in my case, I made sure to have version 11 and 17 installed in my environment, however the problem persists.
Any recommendations on what to check?.

phamhien2102, 2022-07-28:
I also faced this issue (release 8.4.0). My solution:
Firstly, add lines into **node_modules/react-native-ble-manager/android/build.gradle** : 
```compileSdkVersion safeExtGet("compileSdkVersion", 30)
        // Add these lines
	compileOptions {
		sourceCompatibility JavaVersion.VERSION_1_8
		targetCompatibility JavaVersion.VERSION_1_8
	}

	defaultConfig {
		minSdkVersion safeExtGet("minSdkVersion", 21)
	}
```

Finally, I used [patch-package](https://www.npmjs.com/package/patch-package) to patch this library, please run this command in terminal:

`yarn patch-package react-native-ble-manager`

So, hope this help everybody