innoveit/react-native-ble-manager #870: readRSSI value showing RSSI value only once after connection . I want updated RSSI value on each request.

prasadrokade09, 2022-06-16:
async getRSSI(peripheral) {
BleManager.retrieveServices(peripheral.id).then((peripheralData) => {
console.log('Retrieved peripheral services', peripheralData);
BleManager.readRSSI(peripheral.id).then((rssi) => {
console.log('Retrieved actual RSSI value', rssi);
});
});
}
I am using this method to get RSSI value . Currently the code is executing only once when I connected to BLE device. Is that any solution for that or any other method is present in library which gives me updated RSSI value? Please help!!!

marcosinigaglia, 2022-06-16:
Use readRSSI without call retrieveServices before

closed 2022-06-16 by marcosinigaglia

prasadrokade09, 2022-06-16:
Tried without calling retrieveServices showing same issue. The readRSSI method is execute only I disconnect and connect with device  again.

prasadrokade09, 2022-06-17:
Dear sir,
            Thank you for quick reply .Sir I have implemented your solution in my code . Unfortunately it doesn’t work for me . So is there any different solution for get RSSI value. Basically I need updated rssi on button click.

Regards,
Prasad