innoveit/react-native-ble-manager #874: Long Writes (greater than MTU) of Peripheral Characteristics are not working properly (offset is not correct)

phimantaco, 2022-07-01:
**Version**
- react-native-ble-manager v8.0.2
- react-native v0.64.1

**Describe the bug**
I am seeing a problem when I call BLEManager.write() with a write of 200 bytes and maxByteSize of 20.  On the peripheral device side, I am seeing the ten write events with the proper data but the offset value for all of the writes is 0.  So the peripheral peripheral device event does not know what data goes at what offset.  The peripheral device does support Reliable Writes for that characteristic.

**To Reproduce**

**Expected behaviour**
I would expect to see that the offset value for each of the events after the first one being set to the proper value.  I would expect Peripheral.java to pass along the offset as it queues the writes.

Ideally, I would want a Reliable Write to be done with a beginReliableWrite before any of the data is sent and then an executeReliableWrite after all the data is sent.  The peripheral supports this.

I am able to do write properly with the Nordic nRF Connect App.  It creates a queue write with a Prepare Write, Writes, and Execute Write at the end.

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy 20 Ultra
 - OS: Android 12
 - react-native-ble-manager version: 8.0.2 
 - react-native version: 0.64.1

**Additional context**

tientv-hybrid, 2022-07-12:
try using requestMTU: https://github.com/innoveit/react-native-ble-manager#requestmtuperipheralid-mtu-android-only-api-21

closed 2022-07-25 by marcosinigaglia

JRGunthner, 2023-05-17:
I've the same problem. How can I solve it?