innoveit/react-native-ble-manager #877: Illegal callback invocation from native module.

thomasviaud, 2022-07-07:
**Describe the bug**
Hello thanks a lot for this package. Everything works fine but on Android our app is randomly crashing (only Android) due to this unhandled error :

```

Illegal callback invocation from native module. This callback type only permits a single invocation from native code.
```
```
com.facebook.react.bridge.CallbackImpl in invoke at line 26
it.innove.Peripheral$6 in run at line 676
android.os.Handler in handleCallback at line 938
android.os.Handler in dispatchMessage at line 99
android.os.Looper in loop at line 233
android.app.ActivityThread in main at line 8010
java.lang.reflect.Method in invoke
com.android.internal.os.RuntimeInit$MethodAndArgsCaller in run at line 631
com.android.internal.os.ZygoteInit in main at line 978
```

**Smartphone (please complete the following information):**
 - Device: One plus 7T
 - OS: Android 11
 - react-native-ble-manager version: 8.4.0
 - react-native version: 0.67.4

**Additional context**
In our app we might connect to several bluetooth peripherals, disconnect and reconnect from them multiple times. We are ensuring that the bluetooth module is initialized only once. Is it possible that 2 callbacks are sent at the same time causing this error (writing error or connection error) ?

Thanks.

marcosinigaglia, 2022-07-25:
Hi, do you know which callback is called twice?

yousrasd, 2023-02-09:
@thomasviaud look at https://github.com/innoveit/react-native-ble-manager/issues/900. Seems like a fix was done in https://github.com/innoveit/react-native-ble-manager/pull/923 for this.

closed 2023-12-05 by marcosinigaglia