innoveit/react-native-ble-manager #882: NullPointerException: Attempt to invoke interface method

Burzo, 2022-07-25:
**Describe the bug**

The following error is reported in Crashlytics for some of our users:
```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke interface method 'void com.facebook.react.bridge.Callback.invoke(java.lang.Object[])' on a null object reference
       at it.innove.Peripheral$6.run(Peripheral.java:666)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loopOnce(Looper.java:233)
       at android.os.Looper.loop(Looper.java:344)
       at android.app.ActivityThread.main(ActivityThread.java:8204)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:589)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1071)
```

**To Reproduce**
I've never encountered this during development as far as I remember, so I'm not sure how to reproduce this.

**Smartphones that reported this:**
 - Device: Galaxy S21 Ultra 5G, Galaxy S8
 - OS: Android 12 and Android 9
 - react-native-ble-manager version: 8.0.3
 - react-native version: 0.68.1

**Additional context**
I wish I had more information to give, but I'm afraid this is all I know at the moment.
```
at it.innove.Peripheral$6.run(Peripheral.java:666)
```
This line is the only indicator that I noticed.

Hopefully someone can help out.

Burzo, 2022-08-08:
This issue now also occurred during development as well, immediately after scanning stopped, but I can't seem to reproduce it. I got the Android logcat though:

```
08-08 09:49:00.010 16033 16052 I AiAiEcho: Predicting[1]: { Feature:1, creation:2022-08-08T07:37:12.276Z, expiry:2022-08-08T13:37:12.276Z, isExpired: false}
08-08 09:49:00.012 16033 16052 I AiAiEcho: Ranked targets strategy: WEIGHTED, count: 1, ranking metadata: { cardId: WEATHER-569377242, card type: 1, update time: 2022-08-08T07:37:12.276Z, expiration time: 2022-08-08T13:37:12.276Z, ranking metadata: RankingMetaData(urgency=0.966, priority=0.3, recency=1.0, rankScore=0.28980002, importantTimePoints=# mak@7bca4) }
08-08 09:49:00.014 16033 16052 I AiAiEcho: #postPredictionTargets: Sending updates to UISurface lockscreen with targets# 1
08-08 09:49:00.016 16033 16052 I AiAiEcho: #postPredictionTargets: Sending updates to UISurface home with targets# 1
08-08 09:49:00.016 16033 16052 I AiAiEcho: #postPredictionTargets: Sending updates to UISurface media_data_manager with targets# 0
08-08 09:49:00.020 18459 18459 W BcSmartspaceCard: No title view to update
08-08 09:49:00.026  1804  1804 W BcSmartspaceCard: No title view to update
08-08 09:49:00.038 17671 17716 D ReactNativeBleManager: onCharacteristicChanged: 74636C765F726571756573742035313030303030 from peripheral: CB:39:6B:B8:0D:23
08-08 09:49:00.083 17671 17716 D ReactNativeBleManager: onCharacteristicChanged: 300D0A74636C765F726573706F6E736520353130 from peripheral: CB:39:6B:B8:0D:23
08-08 09:49:00.129 17671 17716 D ReactNativeBleManager: onCharacteristicChanged: 303030303430313030303030300D0D0A3E200A from peripheral: CB:39:6B:B8:0D:23
08-08 09:49:00.189 17671 17781 D ReactNativeBleManager: Write without response to: CD:17:25:BA:A8:8E
08-08 09:49:00.189 17671 17781 D ReactNativeBleManager: Message(23): 74636C765F726571756573742035313030303030300D0A
08-08 09:49:00.191 17671 17716 D ReactNativeBleManager: Command queue empty
08-08 09:49:00.201 17671 17716 D ReactNativeBleManager: Command queue empty
08-08 09:49:00.238 17671 17780 I ReactNativeJS: Written tclv_request 51000000 to the device 1e002700-1850-4b35-4338-362000000000 (09:49:00-238)
08-08 09:49:00.299 17671 17716 D ReactNativeBleManager: onCharacteristicChanged: 74636C765F726571756573742035313030303030 from peripheral: CD:17:25:BA:A8:8E
08-08 09:49:00.300 17671 17716 D ReactNativeBleManager: onCharacteristicChanged: 300D0A74636C765F726573706F6E736520353130 from peripheral: CD:17:25:BA:A8:8E
08-08 09:49:00.309   740  2274 W bthal.bqr: ParseLinkQualityRelatedEvt: Appro LSTO , Handle: 0x0043, RSSI_C0: -66, RSSI_C1: -75, TxPw_C0: -128, ReTxPw_C0: -128, TxPw_C1: 9, ReTxPw_C1: 9, BFTx: 0x01, BFReTx: 0x01, DivTx: 0x01, DivReTx: 0x01
08-08 09:49:00.309   740  2274 I bthal.flowcontrol: HciEventPreProcess: HCI Event 0xff Opcode 0 Length 225 unhandled by debug_VendorEvent
08-08 09:49:00.310  1788  2603 W bt_stack: [WARNING:btif_bqr.cc(390)] Appro LSTO , Handle: 0x0043, DH1, Central, PwLv: 9, RSSI: -66, SNR: 0, UnusedCh: 25, UnidealCh: 0, ReTx: 55, NoRX: 55, NAK: 0, FlowOff: 0, OverFlow: 0, UndFlow: 0
08-08 09:49:00.310  1788  2603 I bt_stack: [INFO:btsnoop.cc(338)] clear_l2cap_allowlist: Clearing acceptlist from l2cap channel. conn_handle=67 cid=4:4
08-08 09:49:00.310  1788  2603 W bluetooth: system/bt/stack/gatt/gatt_attr.cc:493 gatt_disc_cmpl_cback: Gatt discovery completed with errors status:133
08-08 09:49:00.310  1788  2603 I bluetooth: system/bt/stack/gap/gap_ble.cc:298 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:09:12 callback:Disconnected
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:3 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:4 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:5 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:6 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:7 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:8 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:9 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:10 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:11 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:12 transport:le reason:GATT_CONN_TIMEOUT
08-08 09:49:00.310  1788  2603 I bt_bta_gattc: system/bt/bta/gatt/bta_gattc_act.cc:1115 bta_gattc_conn_cback: Disconnected att_id:13 transport:le reason:GATT_CONN_TIMEOUT
[... cut ...] imera.GmsIntentOperationService$PersistentTrustedReceiver
08-08 09:49:00.332  1549  1583 W BroadcastQueue: Permission Denial: receiving Intent { act=android.bluetooth.device.action.ACL_DISCONNECTED flg=0x5000010 (has extras) } to com.google.android.apps.wearables.maestro.companion/.connectivity.ClassicBTReceiver requires android.permission.BLUETOOTH_CONNECT due to sender com.android.bluetooth (uid 1002)
08-08 09:49:00.341 20042 20236 I AssistantForeground: (REDACTED) Get launcher package: %s
```

adumat, 2023-04-20:
Have you find a solution for this issue? i have also this issue but with different versions (rn and ble-manager)

closed 2023-12-04 by marcosinigaglia

Burzo, 2024-02-05:
Why was this closed? The issues still exists, and I can see it reliably pop up in our logs.

marcosinigaglia, 2024-02-05:
Hi @Burzo , we only support latest library version. If you have the error with the latest I'll reopen the issue.