innoveit/react-native-ble-manager #886: 100ms delay between packages

ApostolosMavroukakis, 2022-08-04:
Im communicating with a bluetooth device, and im having trouble optimizing performance.
Even though the mtu is about 247 bytes the data is sent in 64bytes, more importantly tho its braking the data to multiple packages that have about 100ms delay between them. For a 300byte json it takes 500ms for the data to reach its destination.

[image]

marcosinigaglia, 2022-09-14:
Hi, probably is a device problem that cap the MTU to 64. The BLE protocol is not designed to transfer data.

ApostolosMavroukakis, 2022-09-14:
> 

yes and yes, it was indeed the device that capped at 64. I dont understand the delay though. Its been tested with native code with passable results as well.

marcosinigaglia, 2022-09-15:
The react native bridge is the bottleneck, are you using writeWithoutResponse or write?

ApostolosMavroukakis, 2022-09-15:
I'm using write

`  const writeData = (stringToWrite) => {
    if (stringToWrite) {

      console.log("writeData function", stringToWrite)

      const data = stringToBytes(stringToWrite);
      BleManager.write(
        peripheralId,
        service,
        characteristicRX,
        data,
        247
      )
        .then(() => {
          console.log("Write: " + stringToWrite);
        })
        .catch((error) => {
          console.log(error);
        });

    }
  }`

ApostolosMavroukakis, 2022-09-15:
Will try and come back to you, thanks a ton

closed 2023-12-04 by marcosinigaglia