innoveit/react-native-ble-manager #892: Hi @Innoveit...

homerotorrijos, 2022-08-12:
**Before open an issue**
- Check the closed issues, your question maybe is not new.
- We can't debug or reproduce hardware issue.
- If the library is not working 99% you did something wrong, in the code, installation or in the phone permissions.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - react-native-ble-manager version: 
 - react-native version:

**Additional context**
Add any other context about the problem here.

pradeep11, 2022-08-16:
Any update on this? I am facing the same issue

Jack-QMY, 2022-08-16:
我也面临一样的问题

thomasviaud, 2022-08-16:
Hello, you also need to ask the BLUETOOTH_SCAN permission at **runtime** for SDK >=31

Jack-QMY, 2022-08-17:
我已经可以正常运行了，但是我在真机上运行的，跑虚拟机会出现这个问题

Jack-QMY, 2022-08-17:
I can already run it normally, but running on a real machine, the virtual machine will have such a problem, my sdk version is 31

marcosinigaglia, 2022-08-17:
You can't use BLE in a virtual machine.

closed 2022-08-17 by marcosinigaglia