innoveit/react-native-ble-manager #900: ERROR Invariant Violation: No callback found with version 8.4.3

fabianbru, 2022-09-02:
**Describe the bug**
If the device is disconnected on write the app crashes with:

Metro logs:
```
Invariant Violation: No callback found with cbID 18325 and callID 9162 for  BleManager.write - most likely the callback was already invoked. Args: '["Device disconnected"]', js engine: hermes
```

LogCat output:
```
FATAL EXCEPTION: main
Process: <redacted:bundleId>, PID: 16050
java.lang.RuntimeException: Illegal callback invocation from native module. This callback type only permits a single invocation from native code.
	at com.facebook.react.bridge.CallbackImpl.invoke(CallbackImpl.java:26)
	at it.innove.Peripheral$6.run(Peripheral.java:681)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7870)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

```

It looks like the issue from https://github.com/innoveit/react-native-ble-manager/issues/885 but on writing data. Or perhaps the issue is not fixed completely? So I added a try/catch block:

JS Code with try/catch:
```js
    try {
      return await this.manager.write(deviceId, serviceUUID, characteristicUUID, valueByteArray, 512)
    } catch (e) {
      console.log('write failed', e)
    }
```

Metro logs with try/catch:
```
write failed Device disconnected
```

**But LogCat output is the same and the app is still crashing.**

**To Reproduce**
1. write data (with response)
2. disconnect device (e.g. by removing battery)

**Expected behavior**
A catchable exception without crash.

**Smartphone (please complete the following information):**
 - Device: Pixel 3a
 - OS: android 12
 - react-native-ble-manager version: 8.4.3
 - react-native version: 0.69.5

ManigandanRaamanathan, 2022-09-06:
facing same issue

kevinswartz, 2022-10-24:
We are on v8.4.3 of this plugin, and I think we're experiencing this error as well.  We cannot reproduce this locally so far, but we have customer reports of this on android 12 and 13.  We don't have very good visibility into the problem beyond this stack trace:
```
java.lang.RuntimeException: Illegal callback invocation from native module. This callback type only permits a single invocation from native code.
  at com.facebook.react.bridge.CallbackImpl.invoke(CallbackImpl.java:25)
  at it.innove.Peripheral$6.run(Peripheral.java:666)
  at android.os.Handler.handleCallback(Handler.java:942)
  at android.os.Handler.dispatchMessage(Handler.java:99)
  at android.os.Looper.loopOnce(Looper.java:201)
  at android.os.Looper.loop(Looper.java:288)
  at android.app.ActivityThread.main(ActivityThread.java:7898)
  at java.lang.reflect.Method.invoke(Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)

```

Koxx3, 2022-11-12:
i have the same issue. i've tried catching this in js & java... impossible for now. the error comes to the UI 😵‍💫

Koxx3, 2022-11-13:
ok, i've think I've found something.
the callback is not reset after sendBackrError() is called.
then the callback is called an second time when the onConnectionStateChange() loop on the callback to clear all pending callbacks.

Koxx3, 2022-11-13:
you can try my fix with :
```
yarn remove react-native-ble-manager
yarn add Koxx3/react-native-ble-manager.git
```
if it works for you too, i'll make a pull request.
it fixed the issue for me with read & write interrupted without BLE device answer.

Koxx3, 2022-11-13:
works great for me ... so I pushed the PR : 
https://github.com/innoveit/react-native-ble-manager/pull/923

traviswsims, 2022-11-14:
> works great for me ... so I pushed the PR : #923

This fixed the issue for me as well.  Thank you!

Koxx3, 2022-11-14:
ahhhh ! cool !

closed 2022-11-15 by marcosinigaglia