innoveit/react-native-ble-manager #905: Device scan on Android does not discover all BLE devices

dominikwagner, 2022-09-17:
Hi

we are using this library to scan for smartlocks and this is working great for ios and Android (Google Pixel 4, Android 13). On ios we are also finding all other nearby BLE devices. 

On Android there are BLE devices which are **not found** with the scan method.
When I start a scan with nrf connect on Android all BLE devices are found.

In short:
- ios: everything is working
- Android: BLE devices is missing when I scan with this library. But it is found when scanning with e.g. nrf connect app or any other BLE scanning app.

Do you have any idea why some devices are missing in the result? Because I checked the nrf connect app I am 100% sure the device is advertising. I also tried different scan modes, match modes, etc.

Thanks

ValentinOsvaldo, 2022-09-24:
I have the same problem, in my case it does not detect my headsets

georgelviv, 2022-11-07:
Same for me (Samsung S20, Android v12, Android API v.31)

azilberbaum, 2022-11-10:
Any solutions?  I am seeing the same thing PIxel 4 on Android 13 works fine, Microsoft DUO on Android 11, finds nothing

dominikwagner, 2022-11-11:
Hi

are you scanning with scan mode LOW_LATENCY?
The scan mode controls how long and with which frequency your device is searching for BLE devices. The default mode is LOW_ENERGY. In the low energy mode Android scans for a very short time (I believe 0.5 seconds) and then pauses for a long time (I believe 4 seconds). Depending on how often the BLE device is advertising it may take a long time on Android to find the BLE device. When you change the scan mode to low latency your device is scanning without interruption and so the BLE device is found faster (but it consumes also way more energy).

But we had another hidden problem:
Android 13 introduced the [neverForLocation](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions) permission flag. If you set this flag on the BLUETOOTH_SCANpermission you can remove the ACCESS_FINE_LOCATION permission for ble scanning (if location is not otherwise needed). In the docs you can find the following note If you include neverForLocation in your android:usesPermissionFlags, some BLE beacons are filtered from the scan result.

We never thought about that because we have not inserted this flag in our manifestfile. But we used a library which included this flag (we found it looking at the merged manifest file....). After removing this flag with tools:remove on the bluetooth scan permission we found the BLE device which was missing in the Android scan results:
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags" />

So if you can find not all BLE devices on Android 13 and you are sure that the missing device is a BLE device than check if ´neverForLocation` is in use (check the merged manifest file - maybe a lib is using it). As written in the docs this could result in missing devices in the scan result.

removing this flag solved the problem for me.

closed 2022-11-11 by dominikwagner

azilberbaum, 2022-11-11:
I am having the opposite problem, everything works fine on Android 13, but Android 11 and 10 does not find any devices :(

dominikwagner, 2022-11-11:
are you sure you are declaring the right permissions? Android 11 and lower needs additional permissions https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

azilberbaum, 2022-11-12:
Found the issue, was using: 
PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION 
instead of 
PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION 

In both the code and the Android manifest