innoveit/react-native-ble-manager #909: Not able to find the bluetooth connected device list.

hupptechnologies, 2022-09-26:
**Mobile Device Environment**
Provide a list of operating systems on which this issue is relevant.
- Device: iPhone11
- OS: 15

**Application Environment**
Provide information about your development environment:
- React Native version: 0.63.4

**Describe the bug**
When i try to scan or try to find the bluetooth connected device then i am always getting empty array even the bluetooth device is connected.

np-singh, 2023-01-31:
@hupptechnologies Were you able to solve this?

Nithinmon-D, 2023-05-25:
any solution ?

amanrajtoppoart007, 2023-09-16:
Same Issue with
React Native - 0.71.12
working good in android device but not able to get the connected peripherals list in IOS ,

`const handleGetConnectedDevices = () => {
    setIsScanning(true);
    BleManager.getBondedPeripherals()
      .then(results => {
        if (results.length === 0) {
          setIsScanning(false);
          console.log('No connected bluetooth devices');
        } else {
          for (let i = 0; i < results.length; i++) {
            let peripheral = results[i];
            peripheral.connected = true;
            peripherals.set(peripheral.id, peripheral);
            setConnectedDevices(Array.from(peripherals.values()));
          }
        }
      })
      .finally(() => {
        setIsScanning(false);
      });
  };`
  
  this is my code

closed 2023-12-24 by marcosinigaglia