innoveit/react-native-ble-manager #910: BleManagerDidUpdateValueForCharacteristic not receiving HEX value after 00 (Null in HEX)

prasadrokade09, 2022-09-29:
Hello there, I am trying get HEX values from ESP32 BLE device as a response in BleManagerDidUpdateValueForCharacteristic listener . Suppose this is my actual response array which is sending from ESP32 **[FF AA FF AA 72 00 BC FF AA FF AA],** but BleManagerDidUpdateValueForCharacteristic  receiving only   **[FF AA FF AA 72]** .After **00** it  terminate the response .I need whole HEX response including 00. Help needed in this!

//Add Listener Code  

bleManagerEmitter.addListener(
      'BleManagerDidUpdateValueForCharacteristic',
      this.handleUpdateValueForCharacteristic,
    );

//Actual Function where I got incomplete value array 

  handleUpdateValueForCharacteristic (data)
 {
        console.log("Value:",data.value);
 }

jath03, 2023-03-30:
I am also running into this issue.  Were you ever able to fix it?  Did you find a good workaround?

jath03, 2023-03-30:
Well, it turns out the values just weren't getting sent from the BLE server side.  I suspect that was the case for the original issue as well, so this can probably be closed.