innoveit/react-native-ble-manager #913: React-native Ble Manager Works on Android But Dont works on IOS Err:could not find service with uuid

huseyinsafaunlu, 2022-10-10:
As I mentioned here https://github.com/innoveit/react-native-ble-manager/issues/912
I do the same things on android it works but on IOS it doesn't work.

- I can scan devices on IOS and Android.
- I can blemanger.connect to any device on IOS and Android..
- **I can't startNotification on my device on IOS but can on android.**
 

I tried this:

I can connect to my device using my variable: connectedDevices.id with is in this case "E8595779-BEF2-FB78-741E-E855465F760C"

**_As seen on the picture below, I connected to device using ID, But it can't find service with UUID "FFE0"
But actually there is service with UUID "FFE0"_**

[![All Info Present Here][4]][4]

  [4]: https://i.stack.imgur.com/TdUYG.jpg

marcosinigaglia, 2022-10-10:
Did you call retrieveServices?

huseyinsafaunlu, 2022-10-10:
@marcosinigaglia Yes when I call the retrieve services I get the values on the picture that at the end of the question. It returns connected peripherals: [{id: "E75....

I will double check it in a min..
 thanks fo answering

huseyinsafaunlu, 2022-10-10:
I did retrieve sending then it **_weirdly_** started sending the data. However I can't recieve anything, I can send data I suppose...

[image]

marcosinigaglia, 2022-10-10:
Did you read the project [note](https://github.com/innoveit/react-native-ble-manager#note)?
You have to retrieve the services data before startNotification or write or read.
Are you using write or writewithoutresponse?

huseyinsafaunlu, 2022-10-10:
Hi again, @marcosinigaglia

closed 2022-10-10 by huseyinsafaunlu

SMModan, 2023-09-23:
> * short

Hi 
Will you able to share the snippet if possible. i am stuck with same issue. 
really appriciate it
Thanks

SMModan, 2023-09-23:
> Hi again, @marcosinigaglia I read the project note now I did retrieve services and I used writeWithoutResponse instead of write. AND IT WORKED. Thanks So much.
> 
> What I learned is if you want ble manager to work on IOS, You should change your working Android code,
> 
> * **Service and char UUID should be short form.**
> * **You should retrieve services after you connect.**
> * **You should use writewithoutresponse.**
>   Much love

Hi 
Will you able to share the snippet if possible. i am stuck with same issue. 
really appriciate it
Thanks