innoveit/react-native-ble-manager #915: Scan/discover new peripherals not working when phone is locked

remirez, 2022-10-27:
Using [a background process](https://github.com/Rapsssito/react-native-background-actions) I am scanning and connecting to new peripherals without issue. However, whenever I lock the phone, although the process still runs fine, it stops discovering new peripherals. Is there a way to bypass this or is this feature blocked by the OS?

 - Device: Zebra TC57
 - OS: Android 11
 - react-native-ble-manager version: 8.0.2
 - react-native version: 0.68.1

rpopovych, 2023-01-22:
Are you using a filter when scanning? For unfiltered scans, scanning is stopped on screen off to save power:
 
https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)

0x13b, 2023-02-25:
The very description is actually given a little bit below in the [documentation](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(android.bluetooth.le.ScanCallback)).
I found that filtering by service name does the trick.

closed 2023-12-18 by marcosinigaglia