innoveit/react-native-ble-manager #918: setLegacy and Android permissions

dan-developer, 2022-11-08:
The error occurs when calling `.scan`, with the blemanager with forceLegacy set to `true`.

`BleManager.start({ showAlert: false , forceLegacy: true})`

The crash log:

`11-08 12:12:18.110  4186  4186 E AndroidRuntime: FATAL EXCEPTION: main
11-08 12:12:18.110  4186  4186 E AndroidRuntime: Process: com.cleanproject, PID: 4186
11-08 12:12:18.110  4186  4186 E AndroidRuntime: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10292, packageName = com.cleanproject, attributionTag = null, token = android.os.BinderProxy@c06ade3, next = null }: AdapterService getRemoteName
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Parcel.createExceptionOrNull(Parcel.java:2438)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Parcel.createException(Parcel.java:2422)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:2405)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Parcel.readException(Parcel.java:2347)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.bluetooth.IBluetooth$Stub$Proxy.getRemoteName(IBluetooth.java:5527)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.bluetooth.BluetoothDevice.getName(BluetoothDevice.java:1913)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at it.innove.LegacyScanManager$1$1.run(LegacyScanManager.java:34)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:938)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:226)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:313)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:8751)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
11-08 12:12:18.110  4186  4186 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)`

tigerjiang, 2022-12-27:
@dan-developer I have meet this question, Did you have fix it?

dan-developer, 2022-12-27:
@tigerjiang after upgrade to the last version of "Android-BLE-Library" the errors has fixed on Android 12.

closed 2022-12-27 by dan-developer