innoveit/react-native-ble-manager #925: Can't disconnect from bluetooth device on android

Betapig1, 2022-11-17:
**Describe the bug**
After successfully connecting to my arduino device, calling BleManager.disconnect does not disconnect the app from the device. I am confirming this in 2 ways, by checking the logs on my arduino which I have confirmed will tell me when a device disconnects, as well as checking BleManager.getConnectedPeripherals

The disconnect is not throwing an error, from the functions point of view it is not having any issues (as in the catch block never runs) however no disconnection ever actually happens.

I've confirmed that the ID I'm sending in is the correct one, and if I sent in an ID that doesn't corrolate to any attatched devices it says "Peripheral not found"

**Expected behavior**
Expected behaivor is for my device to fully disconnect from the bluetooth device when running BleManager.disconnect

**Smartphone (please complete the following information):**
 - Device: Galaxy Z Flip Original
 - OS: Android version 12
 - react-native-ble-manager version: 8.4.4, and 7.6.3
 - react-native version:0.70.5

Here is the code if that helps 
`
const disconnectBluetooth = async () => {
  console.log('DISCONNECTING', id);
  // manager.disconnect(id);
 await BleManager.disconnect(id)
    .then( async(results) => {
      // Success code
      console.log('Disconnected',results);

    })
    .catch(error => {
      // Failure code
      console.log(error);
    });

    var discovered = await BleManager.getConnectedPeripherals([])
    console.log("got end", discovered[0])
     
  
    

};`

jayjo34, 2023-01-25:
Hi ! I had this behavior too and fixed it after a long time of trial and error: 
1. don't use discoverPeripheral Event, it seems like it interfere with the connect method, use `BleManager.getDiscoveredPeripherals` method instead, AFTER(!!) stop scanning
2. Be sure that you are not scanning!
3. connect to device
4. disconnect with force param set to false`BleManager.disconnect(<id>, false)` on Android

Good luck :-)

zhifeng-heng, 2023-06-28:
I am also facing the same issue here on Android. It happens for multiple IoT devices that i am connecting to.

I have also tried all the solutions i can find but still does not resolve it. Anyone can shed some light, please?

evan0204, 2023-08-07:
same issue,  Help!!!!

marcosinigaglia, 2023-12-10:
This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 7 days.

reubenmorgan, 2023-12-14:
Hi,

I am also experiencing the same issue on Android.

I have a list of connected devices, when I try to disconnect from a device all seems fine, the device is no longer shown in the getConnectedPeripherals list and isPeripheralConnected returns false, but the app has not disconnected.  If I close the app that triggers a disconnect as expected.

It seems like the react-native-ble-manager code is removing the device from it's own internal list but is not correctly triggering a disconnect with the underlying Android code.

I have tested this on a number of Android devices and I'm using the current latest version: 11.0.6.  I have tried the disconnect function with force set to both true and false.

Any suggestions what I could try to resolve this?  Currently this is blocking shipping an app to a customer.

Thanks

closed 2024-03-23 by marcosinigaglia